maxgerhardt/platform-raspberrypi #10: Raspberry Pico W

sstaub, 2022-07-17:
Support for Raspberry Pi Pico W.

maxgerhardt, 2022-07-19:
This is currently added but untested by me on real hardware. The https://github.com/maxgerhardt/platform-raspberrypi/tree/develop/examples/arduino-wifi-scan compiles with the latest platform version.

maxgerhardt, 2022-07-21:
Due to a linking bug that will be fixed in [this PR](https://github.com/earlephilhower/arduino-pico/pull/696), you currently have to call the WiFi init function manually, but then it was [reported to work](https://github.com/earlephilhower/arduino-pico/issues/66#issuecomment-1191302275)

maxgerhardt, 2022-07-21:
Fixed in https://github.com/maxgerhardt/platform-raspberrypi/commit/bd2f3e756ba7340150f2bebc2d46110988243018, so after a `pio pkg update -g -p raspberrypi` you should be good to go.

sstaub, 2022-07-22:
Tested with wifi-scan example and works fine.

closed 2022-07-22 by sstaub