maxgerhardt/platform-raspberrypi #11: after last update nothing go on

sstaub, 2022-07-19:
After update to the last version, nothing did work. I get many, many errors. See screenshot:
[image]
The project works fine yesterday.
Also the first rebuild of the Intellisense Index need some minutes on my MacBook Pro 16'' (2019).

maxgerhardt, 2022-07-19:
I've updated the Arduino core for the Pico W support, I'm not sure if those conflic with your Ethernet usage there / clashing headers. CI went through with the basic examples with no problems. In any case, if it it did clash, you can revert to an earlier version with no problems.

Let's try the newest version first though. In the terminal, execute
```
rm -rf ~/.platformio/packages/framework-arduinopico
rm -rf ~/.platformio/platforms/raspberrypi*
```
and build the project again. Does it fail? 

If yes, modify your `platformio.ini` to use the previous core version.
```ini
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
board = pico
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#2.2.2
; if using picoprobe SWD upload / debugging
;upload_protocol = picoprobe
;debug_tool = picoprobe
```
I **highly** recommend pinning platform and framework packages versions once you find a working config. This is bleeding edge staff, and PIO staff deletes the older version of packages once newer versions are uploaded (hence I had to hotfix the used compiler and framework versions a few times).

sstaub, 2022-07-19:
It is a general problem. I made an issue on @earlephilhower #687

maxgerhardt, 2022-07-19:
Then this is tracked in https://github.com/earlephilhower/arduino-pico/issues/687 further.

For stability, I also recommend you set `platform = https://github.com/maxgerhardt/platform-raspberrypi.git#81e537a86e55f4f591e9062f914d9be5c7c7200d` as a known-working version.

closed 2022-07-19 by maxgerhardt