maxgerhardt/platform-raspberrypi #17: Bundled OpenOCD does not handle CMSIS-DAPv2

rgrr, 2022-12-03:
Hello Max,

I'm currently experimenting with CMSIS-DAPv2 for the picoprobe.  During my experiments I found out, that the bundled (Windows) OpenOCD does not handle CMSIS-DAPv2.

Log should look like this:
```
Open On-Chip Debugger 0.11.0-g228ede43d-dirty (2022-11-14-09:09)
...
Debug: 25 3 configuration.c:97 find_file(): found ./cmsis-dap.cfg
Debug: 26 4 command.c:146 script_debug(): command - adapter driver cmsis-dap
Debug: 28 4 command.c:146 script_debug(): command - cmsis_dap_vid_pid 0x2e8a 0x000c 0x2e8a 0x000d 0xcafe 0x4005
...
Debug: 165 47 cmsis_dap_usb_bulk.c:156 cmsis_dap_usb_open(): found product string of 0x2e8a:0x000d 'Picoprobe CMSIS-DAP v2'
Debug: 166 47 cmsis_dap_usb_bulk.c:176 cmsis_dap_usb_open(): enumerating interfaces of 0x2e8a:0x000d
Debug: 167 53 cmsis_dap_usb_bulk.c:223 cmsis_dap_usb_open(): found interface 0 string 'Picoprobe CMSIS-DAP v2'
Info : 168 53 cmsis_dap_usb_bulk.c:324 cmsis_dap_usb_open(): Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000d, serial=E6614103E7176A23
Info : 169 54 cmsis_dap.c:837 cmsis_dap_get_caps_info(): CMSIS-DAP: SWD  Supported
Info : 170 55 cmsis_dap.c:817 cmsis_dap_get_version_info(): CMSIS-DAP: FW Version = 2.0.0
Info : 171 56 cmsis_dap.c:940 cmsis_dap_swd_open(): CMSIS-DAP: Interface Initialised (SWD)
Debug: 172 58 cmsis_dap.c:1012 cmsis_dap_init(): CMSIS-DAP: Packet Size = 512
Debug: 173 58 cmsis_dap.c:1025 cmsis_dap_init(): CMSIS-DAP: Packet Count = 8
Debug: 174 58 cmsis_dap.c:1028 cmsis_dap_init(): Allocating FIFO for 3 pending packets
Info : 175 59 cmsis_dap.c:852 cmsis_dap_get_status(): SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Error: 176 60 cmsis_dap.c:368 cmsis_dap_cmd_DAP_SWJ_Clock(): CMSIS-DAP command CMD_DAP_SWJ_CLOCK failed.
...
```

But the bundled version shows:
```
Open On-Chip Debugger 0.11.0-g610f137 (2022-02-11-13:57)
...
Debug: 25 3 configuration.c:97 find_file(): found ./cmsis-dap.cfg
Debug: 26 4 command.c:146 script_debug(): command - adapter driver cmsis-dap
Debug: 28 4 command.c:146 script_debug(): command - cmsis_dap_vid_pid 0x2e8a 0x000c 0x2e8a 0x000d 0xcafe 0x4005
...
Error: 165 309 cmsis_dap.c:260 cmsis_dap_open(): unable to find a matching CMSIS-DAP device
...
```

Any chances to include an updated version which is fixed?

Sorry if this is the wrong place, but I have to admit that I do not understand where are the actual packet sources came from and you are the closest pointer...

maxgerhardt, 2022-12-18:
The OpenOCD version comes from https://github.com/earlephilhower/pico-quick-toolchain/, can you ask there to activate the CMSIS-DAP compile options?

maxgerhardt, 2022-12-24:
Fixed per https://github.com/earlephilhower/pico-quick-toolchain/issues/23

commit 39b2ae97 (2021-08-18): Update Arduino-mbed package to v2.4.1

Resolve #17
closed 2022-12-24 by maxgerhardt