maxgerhardt/platform-raspberrypi #18: Running from rpi 4b install

MadTooler, 2022-12-18:
Howdy.

I have moved my platformIO project from a win10 machine to a raspberry pi 4b.  I get a compile error that the resources for the arm processor are not found.  Is this project functional on the pi 4b just yet?

Here is my platform.ini :

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
upload_protocol = picotool
platform_packages =
	toolchain-gccarm;noneeabi@~1.90301.0
lib_deps = 
	teemuatlut/TMCStepper@^0.7.3
monitor_speed = 2304000
upload_port = /dev/ttyACM1
```
This is what I get at compile...

```
 *  Executing task in folder serialFirst_h_dualTry: platformio run 

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pico; framework: arduino)
---------------------------------------------------------------------------------------------------------------
Tool Manager: Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714
Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714' requirements for your system 'linux_armv7l'

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

maxgerhardt, 2022-12-18:
>Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714' requirements for your system 'linux_armv7l'

Running a 32-bit OS on a 64-bit Raspberry Pi is currently not supported. Duplicate of https://github.com/earlephilhower/pico-quick-toolchain/issues/22.

You can fix this by putting a 64-bit OS on your SD card.

MadTooler, 2022-12-20:
Thank you for the details.

I am running several other applications on the pi that were problematic with the 64-bit OS.  I can try another fresh install to see if things work better now, but I may be out of luck for a while. 

The duplicate issue noted was marked as completed a couple of weeks ago.  I do not see any code changes.  I assume 32-bit will be possible when a new push is eventually released.  

Can you please confirm if my platform.ini items above the lib-deps are all needed and most current?

Thanks again.

maxgerhardt, 2022-12-20:
```ini
platform_packages =
	toolchain-gccarm;noneeabi@~1.90301.0
```
I don't know where you got that from but that's the wrong syntax. It would have been `toolchain-rp2040-earlephilhower@<version>`. But, that still wouldn't solve your problem, since that tries to source it from the registry, which doesn't have an ARMv7l version, or at least, marked as such ([source](https://registry.platformio.org/tools/earlephilhower/toolchain-rp2040-earlephilhower/compatibility)).

The talk in the referenced issue says the armv6l build is actually binary compatible with the armv7l. Do the following in a shell on the raspbi:

```sh
# "sudo apt install wget" if needed..
wget https://dl.registry.platformio.org/download/earlephilhower/tool/toolchain-rp2040-earlephilhower/5.100300.220714/toolchain-rp2040-earlephilhower-linux_armv6l-5.100300.220714.tar.gz
tar xfv toolchain-rp2040-earlephilhower-linux_armv6l-5.100300.220714.tar.gz
cd toolchain-rp2040-earlephilhower-linux_armv6l-5.100300.220714
cd bin
ldd arm-none-eabi-gcc
./arm-none-eabi-gcc --version
```
If the binary is executable, GCC should output its version.

From there you can rename + move the `toolchain-rp2040-earlephilhower-linux_armv6l-5.100300.220714` folder into `~/.platformio/packages/toolchain-rp2040-earlephilhower`. Edit the `package.json` therein to change `linux_armv6l` to `linux_armv7l`. Then retry building the PlatformIO project -- with your previous `platform_packages` directive deleted of course.

NotFastEnuf, 2022-12-21:
Thank you for the detailed response to this issue.  Trying to follow the procedure outlined above for a 32bit OS but step one is failing for me with usc1.contabostorage.com host address is unable to resolve.  

Following the chain of closed issues, it looks like all the changes to support 32bit os are flagged as already merged a few weeks ago, and I saw mention that we are just waiting on the next build to be uploaded.  I am not sure if that has occured yet, or if maybe there is still something amiss.  Is there anpther approach I can take to get a copy of the toolchain since the wget command will not resolve?

MadTooler, 2022-12-21:
> 

Thanks for the instructions.  My system is isolated from internet for bit.  As soon as I can allow it access again, I will give it try and let you know if it works.

maxgerhardt, 2022-12-21:
>but step one is failing for me with usc1.contabostorage.com host address is unable to resolve.

Well..

```
$ nslookup usc1.contabostorage.com 
Server:		127.0.0.53
Address:	127.0.0.53#53

Non-authoritative answer:
Name:	usc1.contabostorage.com
Address: 209.126.15.85
Name:	usc1.contabostorage.com
Address: 209.126.15.84
```

Alright, then just download it from the [repo releases](https://github.com/earlephilhower/pico-quick-toolchain/releases) instead of via the registry: 

https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.4.0-c/arm-linux-gnueabihf.arm-none-eabi-0196c06.220714.tar.gz

NotFastEnuf, 2022-12-21:
Thank you for your response.  Was able to download from your link.  GCC was able to output the version.  Changed the package.json system to linux_armv7l.  Renamed the folder to toolchain-rp2040-earlephilhower and moved it into .platformio/packages directory.  Unfortunately platformio does not seem to be picking it up.  Same error message that it could not find the package for linux_armv7l.

I understand that this is not the appropriate venue for support... just wanted to report back that I attempted to follow your procedure and express appreciation for your time and efforts.

maxgerhardt, 2022-12-21:
Does ~.platformio/platforms/raspberry/platform.json exist? If yes, edit the version field for the toolchain earlephilhower package from it's original ~1.10.... something string to symlink:// plus the absolute path of the extracted toolchain (folder where package.json is). This path should not be in . platformio/packages itself, use the original download folder.E.g., symlink:///home/pi/Downloads/picotoolchain/(Yes 3 / are correct)

NotFastEnuf, 2022-12-21:
No, I do not have a folder labeled raspberry in  ~.platformio/platforms directory.

MadTooler, 2022-12-22:
I tried this several times and only made a mess.  I did not end up with the same folders or files being generated that you noted.  I will have to try again if I am able to make all my other programs happy under the 64 bit.  Thanks anyway.

NotFastEnuf, 2022-12-22:
I have both a "raspberrypi" folder and a folder named "raspberrypi@src-ff76a3915224135aafad379817f41edd".  Only the second one included any reference to the new toolchain.  I replaced it's version with symlink:// + local path  and after a long intellisense rebuild, my project will compile.  Thank you for the support.  I also had to try a few times and was not successful until starting over with a fresh OS flash and install of vscode & platformio.

Unfortunately though when trying to upload, I am met with the origional error again regarding linux_armv7l.  Of course it is possible to manually place the uf2 file in the pico and this seemed to work on the pi4.  In an attempt to simplify my experience, I tried just moving over to a windows 10 machine, held boot, replaced the RPI2 Boot (Interface 1) drivers with Zadig to libusb-win32, verified it was picked up in device manager, probed picotool.exe info but it did not find a rp2040 device in BOOTSEL mode.   Upload failed with error 1, but I think this is better discussed elsewhere.

maxgerhardt, 2022-12-22:
>my project will compile

Glad to hear compilation is working.

>Unfortunately though when trying to upload, I am met with the origional error again regarding linux_armv7l.

Same game for `tool-openocd-rp2040-earlephilhower` and `tool-mklittlefs-rp2040-earlephilhower`:
* https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.4.0-c/arm-linux-gnueabihf.openocd-e3428fadb.220714.tar.gz
* https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.4.0-c/arm-linux-gnueabihf.mklittlefs-affa497.220714.tar.gz

Download it, extract it somewhere, change `linux_armv6l` to `linux_armv7l` in the `package.json`, reference path of extract tool in the `platform.json` with `symlink://<path>`.

I'll ping Earle if the packages are already submitted and just waiting for manual review from the PIO team, then this problem should go away.

>I tried this several times and only made a mess. I did not end up with the same folders or files being generated that you noted. I will have to try again if I am able to make all my other programs happy under the 64 bit. Thanks anyway.

You might want to try to get a clean PlatformIO install, aka 
```
rm -rf ~/.platformio/packages/*
rm -rf ~/.platformio/platforms/*
```
then clone the platform directly
```
git clone https://github.com/maxgerhardt/platform-raspberrypi.git ~/.platformio/platforms/raspberrypi
```

Then do the package injections for the toolchain, openocd and mklittlefs per above, and create a new project with
```
pio init -b rpipico --ide=vscode 
```
in some folder. It should pick the already installed version up and let you add a `src/main.cpp` and compile it.

maxgerhardt, 2022-12-24:
The new toolchain has been uploaded and is already in the registry.

The easiest way is to remove PlatformIO + esphome cleanly, then reinstall it, and you shall have success in compilation + uploading. See https://registry.platformio.org/tools/platformio/framework-zephyr/versions

NotFastEnuf, 2022-12-25:
Just in time for Christmas, my son will be thrilled!  Thank you for all
your work on this.

MadTooler, 2022-12-29:
> 

Sweet xmas gift!  So far so good!  Good thing too, 64 bit OS is not an option for now due to other conflicts.  

I did the cache and other scrubbing noting in the other issue.  As long as it doesn't give me any headache when I start testing the multiple processors, I think I am up and running.

Thanks soo much and have a great new year!

maxgerhardt, 2022-12-29:
Great, with two confirmed successes I think it's safe to close the issue now. Please reopen if new issues arrise.

closed 2022-12-29 by maxgerhardt