maxgerhardt/platform-raspberrypi #19: Best Practice for Platformio.ini

savejeff, 2022-12-26:
Marry Christmas,

I was just adding Pico W support to my project and was going over my config
currently this is how it looks like

```

[env:BOARD_RP2040_PICO]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
board = pico
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m

```

about the `platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git`. i added this when there where the malloc problems with the newlib code and this gave me the latest release from earlephilhower that fixed the problem. Now that this is also published here, does it still make a difference? On the average, how much does the platformio release lag behind the quite regular releases of earlephilhower?

another small this: it took me some time to find the board name "rpipicow". i think it would be good to update/extent/create a documentation either on https://docs.platformio.org/ or on https://arduino-pico.readthedocs.io/.

I don't know if it's just me or is the Mbed core dead or at least very very slowly kept up to date. 
When comparing the board list on the mbed and earlephilhower give a pretty strong indication. 
i know about the hole discussion as i followed it from the start, but at some point i think it would be best for the community to just switch to the more reliable and better supported core (even if its not the official)
Are there considerations on replacing the official mbed core with the earlephilhower  under https://docs.platformio.org/en/latest/platforms/raspberrypi.html ?

As i found myself looking on quite a lot of different information sources: For somebody finding this searching for Infos, here are the current links I use to find infos and my current build config

```

; platform RP2040 / Raspberry Pi Pico config
; Arduino-Pico Wiki: https://arduino-pico.readthedocs.io/en/latest/index.html
; https://arduino-pico.readthedocs.io/en/latest/platformio.html#what-is-platformio
; https://arduino-pico.readthedocs.io/en/latest/platformio.html#selecting-the-new-core
; Earlephilhower Arduino Core: https://github.com/earlephilhower/arduino-pico
; PlatformIO Core for Alt Core: https://github.com/maxgerhardt/platform-raspberrypi
; Board List: https://github.com/maxgerhardt/platform-raspberrypi/tree/develop/boards
[cfg_rp2040_alt] ; https://docs.platformio.org/en/latest/platforms/raspberrypi.html
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
board = pico
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m

build_flags = 

	-DPIO_FRAMEWORK_ARDUINO_ENABLE_EXCEPTIONS ; Enable Exceptions
	-fstack-protector ; Enable Stack Protector
	;-DPIO_FRAMEWORK_ARDUINO_ENABLE_RTTI ; Enable RTTI

	;-DUSE_TINYUSB ; Adafruit TinyUSB
	;-DPIO_FRAMEWORK_ARDUINO_NO_USB ; No USB stack
	
	;-DPIO_FRAMEWORK_ARDUINO_ENABLE_IPV6 ; IPv6

```

Riffer, 2023-04-08:
I just found out that the installation by package_platform is now deprecated.

You will find the best starting point here:
https://arduino-pico.readthedocs.io/en/latest/platformio.html

savejeff, 2023-04-10:
i dont think you are correct there. 

i you done use the platform_packages, u will get an pretty old version of the arduino pico core as the last release from max gerhard is over six month. it does not support wifi etc

mattordev, 2025-10-03:
I know this is probably a very dead thread by this point, but would this setup (or something similar) work for the Pico W? I can't seem to find a reliable setup that isn't causing me a headache.

earlephilhower, 2025-10-03:
Max has documented everything here: https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst

mattordev, 2025-10-03:
thank you, I have skimmed through it already, and have previously been using his setup [here](https://community.platformio.org/t/what-is-the-definitive-platformio-ini-configuration-for-the-pico-w/40478/2?u=mattordev) - but I'm just getting constant problems even on that one! My poor brain hurts lol.

Here's the thread I've funnily enough just posted re the toolchain issues, I bet the problem is staring me in the face... 
https://community.platformio.org/t/toolchain-errors-with-pico-w-fails-to-build/52688