maxgerhardt/platform-raspberrypi #23: RPi Pico W build fails when using api services in ESPHome

THorst92, 2023-02-20:
Hi,

I am not sure if this might be also a problem for this repository.
I found an issue while using the RPi Pico in ESPHome with the api services and created the following bug: esphome/issues#4160
Since the platformio package is mentioned in the failure messages, I thought to post it also here.

Could anyone maybe help me?

maxgerhardt, 2023-02-22:
This is a fault in ESPHome sources, probably a missing implementation for the RP2040, just like https://community.platformio.org/t/error-ld-returned-1-exit-status-remote-transmitter/32101.

Aka, not a problem of this repo.

I can only advise you to look at the source code you're building and where that
```cpp
long esphome::api::get_execute_arg_value<long>(esphome::api::ExecuteServiceArgument const&)
```
function is that you're missing.

THorst92, 2023-02-22:
Thanks for the hint! I think I got the problem and updated esphome/issues#4160.

closed 2023-02-22 by THorst92