maxgerhardt/platform-raspberrypi #24: ADCInput missing?!

sreimchen, 2023-02-27:
Hi Max,

im using this repo with PlatformIO. But anyhow it is missing the ADCInput.h that is found in Earle's core see here: https://github.com/earlephilhower/arduino-pico/tree/master/libraries/ADCInput

I get an include error and didn't find the corresponding header in the core's files:
[image]
[image]

Do you have any idea? Any help is very much appreciated because I'm currently depending on this to work...

maxgerhardt, 2023-02-28:
Yes, the core reference is not always up to date since I want to make it compiles and works file when I update. Will be updated soon after I verify bluetooth examples from SDK 1.5 upgrade work.

sreimchen, 2023-02-28:
That makes sense, thank you for your work by the way :)

maxgerhardt, 2023-03-06:
Arduino-Pico was updated in 090343e4478f749da5347487ca42c231df75aab3.