maxgerhardt/platform-raspberrypi #25: Installation doesn't work - Filenames too long?

MrTucks, 2023-03-04:
Steps to reproduce:
0. This platform is not installed in Platformio. (I had tried to install it before. But I uninstalled it in the 
1. Create new Platformio Project using the Raspberry Pi RP2040 platform (the official one)
2. Project gets created successfully
3. change platformio.ini to
```
[env]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m

[env:pico]
board = pico
; if using picoprobe SWD upload / debugging
;upload_protocol = picoprobe
;debug_tool = picoprobe
```
5. wait for platformio to (try to) install everything and look at the output. There are a few errors at the end and it doesn't seem to install correctly.

If I then click build (in in platformio project tasks) it tries to do the same thing, I think. But the output is a bit more verbose. Same error though.

Output after changing platformio.ini:
[output2.txt](https://github.com/maxgerhardt/platform-raspberrypi/files/10889403/output2.txt)
```
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long
[...]
Submodule path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
fatal: Unable to checkout '17761f5cf9dbbf2dcf665b7c04934188add20082' in submodule path 'pico-sdk/lib/tinyusb/hw/mcu/wch/ch32v307'

fatal: Failed to recurse into submodule path 'pico-sdk/lib/tinyusb'

Submodule path 'tools/libbearssl/bearssl': checked out 'f294aa022f82833941ffdf3b0f306c2072e3a26c'
Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
fatal: Failed to recurse into submodule path 'pico-sdk'

Error: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\linus\\.platformio\\.cache\\tmp\\pkg-installing-i_j1ir8r']
```

Output after clicking build:
[output_build.txt](https://github.com/maxgerhardt/platform-raspberrypi/files/10889424/output_build.txt)

Edition	Windows 11 Pro
Version	22H2
Build 22621.1265

I have never used platformio. I'm verry sorry if I've overlooked something obvious, done something wrong, etc.

maxgerhardt, 2023-03-05:
Enable [Win32 LongPaths](https://www.microfocus.com/documentation/filr/filr-4/filr-desktop/t47bx2ogpfz7.html) and retry.

Does it still fail?

MrTucks, 2023-03-05:
Thank you so much! I got it working!

I activated it like your link says and then did a full restart of the PC. Unfortunately, the same error still occurred.
Then I followed this: https://stackoverflow.com/questions/22575662/filename-too-long-in-git-for-windows
`git config --system core.longpaths true`
I tried it again and there was some error about a missing build script, so I just deleted .platformio, reinstalled platformio and tried again.
And it worked!
It installed everything. After changing the board to rpipicow in platformio.ini the LED blinking example just worked on my Pico W!
🥳 

Windows long paths support is kinda wonky, but I guess I can live with it. :D

I think I still have some (minor) issues:
1. Sorry, I forgot to copy and paste the output. But during the installation (after changing the platformio.ini to use this platform) there were some warnings about could not rm some folders because they weren't empty. I can reinstall everything again and get you the log if you want. But I guess since it works, it wasn't that important. 
2. In PIO home, the project inspection doesn't work. 
It gives the following error:
[project inspection.txt](https://github.com/maxgerhardt/platform-raspberrypi/files/10892123/project.inspection.txt)
Platform Tasks -> Advanced -> Verbose Check outputs the following:
[verbose check.txt](https://github.com/maxgerhardt/platform-raspberrypi/files/10892127/verbose.check.txt)

maxgerhardt, 2023-03-05:
>Please note: check results might not be valid!
Try adding --skip-packages

Yes, cppcheck has problems when it's let loose on the toolchain folders. But it's not strictly needed when it's only supposed to check your firmware code. As [per documentation](https://docs.platformio.org/en/latest/projectconf/sections/env/options/check/check_skip_packages.html), add `check_skip_packages = yes` to the `platformio.ini` and retry.

MrTucks, 2023-03-05:
Thank you so much!

closed 2023-03-05 by maxgerhardt