maxgerhardt/platform-raspberrypi #26: Can no longer resolve dependencies (Error: VCS: Could not process command ...)

minisbett, 2023-03-24:
I have this platform referenced via
```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
```
When opening the project, PIO starts cloning submodules for around 10 minutes. After the submodule pyserial, it gives me an error.
```
Cloning into 'C:/Users/Niklas/.platformio/.cache/tmp/pkg-installing-5z0ah8vy/pico-sdk/lib/tinyusb/tools/uf2/hidapi'...

Submodule path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
fatal: Failed to recurse into submodule path 'pico-sdk/lib/tinyusb'

fatal: Failed to recurse into submodule path 'pico-sdk'

Submodule path 'tools/libbearssl/bearssl': checked out 'f294aa022f82833941ffdf3b0f306c2072e3a26c'
Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
Error: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\Niklas\\.platformio\\.cache\\tmp\\pkg-installing-5z0ah8vy']
```

I have tried re-installing PIO, deleting the .platformio folder, re-installing VSC etc. multiple times without any success. This behavior was replicatable by other people. When executing the exact command via the cmd, everything clones fine without any errors. First I thought this is because I am using a fork of this repository but I tried using this instead and the issue still existed. This started happening after updating to the newest state of the master branch of this repo. Before, I was using a ~2 months behind fork that worked without any issues. What's also mentionable is that the error comes after the pyserial submodule, which is actually the last module which I found out when cloning the repo manually.

minisbett, 2023-03-24:
I've gotten two errors about filenames being too long some log entries before the fatal errors happened, I'm not sure if they are related:
```
From https://github.com/STMicroelectronics/stm32u5xx_hal_driver
 * branch            2e1d4cdb386e33391cb261dfff4fefa92e4aa35a -> FETCH_HEAD
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32u5xx_hal_driver': checked out '2e1d4cdb386e33391cb261dfff4fefa92e4aa35a'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32wbxx_hal_driver': checked out '2c5f06638be516c1b772f768456ba637f077bac8'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/ti': checked out '143ed6cc20a7615d042b03b21e070197d473e6e5'
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long
fatal: Unable to checkout '17761f5cf9dbbf2dcf665b7c04934188add20082' in submodule path 'pico-sdk/lib/tinyusb/hw/mcu/wch/ch32v307'
remote: Enumerating objects: 22533, done.
remote: Counting objects: 100% (1201/1201), done.
remote: Total 22533 (delta 1201), reused 1201 (delta 1201), pack-reused 21332
Receiving objects: 100% (22533/22533), 15.89 MiB | 10.30 MiB/s, done.
Resolving deltas: 100% (20465/20465), completed with 125 local objects.
From https://github.com/ARM-software/CMSIS_5
 * branch                20285262657d1b482d132d20d755c8c330d55c1f -> FETCH_HEAD
Submodule path 'pico-sdk/lib/tinyusb/lib/CMSIS_5': checked out '20285262657d1b482d132d20d755c8c330d55c1f'
Submodule path 'pico-sdk/lib/tinyusb/lib/FreeRTOS-Kernel': checked out '2a604f4a2818b8354b5e1a39e388eb5e16cfbc1f'
remote: Enumerating objects: 156, done.
remote: Counting objects: 100% (95/95), done.
Receiving objects:  62% (97/156)remote: Total 156 (delta 94), reused 94 (delta 94), pack-reused 61
Receiving objects: 100% (156/156), 16.72 KiB | 5.57 MiB/s, done.
Resolving deltas: 100% (123/123), completed with 38 local objects.
From https://github.com/lwip-tcpip/lwip
 * branch              159e31b689577dbf69cf0683bbaffbd71fa5ee10 -> FETCH_HEAD
Submodule path 'pico-sdk/lib/tinyusb/lib/lwip': checked out '159e31b689577dbf69cf0683bbaffbd71fa5ee10'
Submodule path 'pico-sdk/lib/tinyusb/lib/sct_neopixel': checked out 'e73e04ca63495672d955f9268e003cffe168fcd8'
Submodule path 'pico-sdk/lib/tinyusb/tools/uf2': checked out '19615407727073e36d81bf239c52108ba92e7660'
Submodule 'hidapi' (https://github.com/signal11/hidapi) registered for path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi'
Cloning into 'C:/Users/Niklas/.platformio/.cache/tmp/pkg-installing-zyey64xh/pico-sdk/lib/tinyusb/tools/uf2/hidapi'...
remote: Enumerating objects: 1757, done.        
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 1757 (delta 0), reused 4 (delta 0), pack-reused 1753        
Receiving objects: 100% (1757/1757), 2.67 MiB | 2.02 MiB/s, done.
Resolving deltas: 100% (1018/1018), done.
Submodule path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
fatal: Failed to recurse into submodule path 'pico-sdk/lib/tinyusb'
fatal: Failed to recurse into submodule path 'pico-sdk'
remote: Enumerating objects: 12, done.
remote: Counting objects: 100% (12/12), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 7 (delta 5), reused 7 (delta 5), pack-reused 0
Unpacking objects: 100% (7/7), 1.13 KiB | 25.00 KiB/s, done.
From https://github.com/earlephilhower/bearssl-esp8266
 * branch            f294aa022f82833941ffdf3b0f306c2072e3a26c -> FETCH_HEAD
Submodule path 'tools/libbearssl/bearssl': checked out 'f294aa022f82833941ffdf3b0f306c2072e3a26c'
Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/minipadkb/arduino-pico.git', 'C:\\Users\\Niklas\\.platformio\\.cache\\tmp\\pkg-installing-zyey64xh']
```

minisbett, 2023-03-24:
After finding https://github.com/hathach/tinyusb/issues/1939 it seems like this repo needs to update the tinyusb submodule. @maxgerhardt would that be doable?

maxgerhardt, 2023-03-24:
No. The repo points at Arduino-Pico. That the points at the stable Pico-SDK. That points to a TinyUSB version that still has all the submodules.The change will only trickle through after a new Pico-SDK version is released that points to new TinyUSB version and Arduino-Pico repoints at the new Pico-SDK. There's nothing we can do in this repo.

minisbett, 2023-03-24:
Oh right I kind of messed up which repo we are on. Until earlephilhower updates tinyusb the framework won't be useable I gueses. Hm. Will try to implement the changes on a fork so I can work with it for the time being.

closed 2023-03-24 by minisbett

PabloOyarzo, 2023-06-05:
Given the issues from PlatformIO to accept the PR is there a workaround on this? I have the same issue

maxgerhardt, 2023-06-05:
Dependency downloading is completely unrelated to the PR. It doesn't matter a single bit if they accept the PR in platform-raspberry or not here. 

The depenency downloads should work fine. If you're on Windows, make sure that you have enough space (TinyUSB takes up lots of space, have at least 3, if not better 5 GB free!) and make sure you followed https://arduino-pico.readthedocs.io/en/latest/platformio.html#important-steps-for-windows-users-before-installing