maxgerhardt/platform-raspberrypi #32: Issue with using FreeRTOS

aloysiousBenoy, 2023-06-09:
Hey @maxgerhardt thanks for the great work done here. 

I'm trying to use FreeRTOS in my project but I'm unable to access FreeRTOS in my project.
I expect this to be some configuration issue but I can't find why this happening to save my life.  
 I imported ```FreeRTOS.h``` as per the arduino-pico documentation. But I'm unable to access functions like vTaskDelay or xTaskCreate in my code.

This is the error I get: 
[image]

Am I missing something here? As per the arduino-pico documentation, there is out the box support for FreeRTOS and It mentions that all I need to do is just add the header.
Thanks again.

maxgerhardt, 2023-06-09:
Can you post your current `platformio.ini`?

aloysiousBenoy, 2023-06-09:
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
upload_protocol = cmsis-dap
debug_tool = cmsis-dap
lib_deps = 
	seeed-studio/Accelerometer_MMA7660@^1.0.0
	olikraus/U8g2@^2.34.18
	
```

maxgerhardt, 2023-06-09:
Like in traditional FreeRTOS programming, you **have to** `#include <task.h>` to get access to the task functionality.

I've verified that
```cpp
#include <Arduino.h>
#include <FreeRTOS.h>
#include <task.h>
#include <MMA7660.h>
#include <U8g2lib.h>

TaskHandle_t myTaskHandle;
void my_task(void* arg) {
    (void) arg;
    while(1) {
        Serial.println("Hello from my task!");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    vTaskDelete(NULL); // never reached but good practice
}

void setup() {
    Serial.begin(115200);
    xTaskCreate(&my_task, "MYTASK", 1024 /* num of 32-bit stack elements */, NULL, configMAX_PRIORITIES - 2, &myTaskHandle);
}

void loop() { delay(1000); }

// no setup1(), loop1() functions in this case to use Core1
```

compiles normally and does produce the expected serial output.

closed 2023-06-09 by aloysiousBenoy