maxgerhardt/platform-raspberrypi #33: Issues with Auto port selection

PabloOyarzo, 2023-06-23:
I tried it once and it worked, and while trying to run some examples of adafruit dual host usb it broke again.

I tried transfer the .elf file once to see if it was gonna work after that automatically but it didn't and also tried to create the project fresh new and it didnt work either.

**platform.ini**
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
upload_protocol = mbed

[image]

Any guidance?

maxgerhardt, 2023-06-25:
Mhh when you set `upload_protocol = mbed` you need to set `upload_port = F:\` or whatever the upload bootloader disk will show up as.

PabloOyarzo, 2023-06-26:
It worked if I inserted the pico with the BOOT SEL button pressed. Its not forcing reset. 

[image]

PabloOyarzo, 2023-06-26:
Picotool did managed to reset the connection but failed to upload the file

[image]

maxgerhardt, 2023-06-27:
If it fails in picotool, then you don't hve the right drivers installed with Zadig. Please checkout https://github.com/platformio/platform-raspberrypi/issues/2#issuecomment-828586398 and use either libusb or WinUSB drivers.