maxgerhardt/platform-raspberrypi #36: Could not initialize project

taunoe, 2023-08-29:
When I great project with a PIO Project Wizard I get this error:

PIO Core Call Error: "The following files/directories have been created in /home/taunoerik/Documents/PlatformIO/Projects/xiao-rp2040-test\ninclude - Put project header files here\nlib - Put project specific (private) libraries here\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nResolving seeed_xiao_rp2040 dependencies...\n\n\nUnknownBoard: Unknown board ID 'seeed_xiao_rp2040'"

Version:1.9.0+sha.ac45b10

maxgerhardt, 2023-09-03:
Make sure you're using `platform = https://github.com/maxgerhardt/platform-raspberrypi.git`. If that is already the case, do 
```
rm -rf ~/.platformio/platforms/raspberrypi*
```

taunoe, 2023-09-03:
It works when I add it to the .ini file.

But GUI > New project produces this error. Earlier, maybe half a year ago it worked. After I added this repository using Advanced Installation.

closed 2023-10-05 by maxgerhardt