maxgerhardt/platform-raspberrypi #39: Debugging with J-Link on macOS not working.

traxanos, 2023-09-29:
Hi Max,

we found a problem. When we try to start the debugger using J-Link on macOS, we always get the following error:

```Listening on TCP/IP port 2331
Connecting to target...
Connected to target
Waiting for GDB connection...
Reading symbols from /Users/XXXX/Projekte/OpenKNX/XXX/.pio/build/develop_RP2040_JLINK/firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.

PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...
Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x10003A40 (Data = 0x4A391AD3)
Reading register (MSP = 0x20041F00)
Reading register (PSP = 0xFFFFFFFC)
Reading 64 bytes @ address 0x20041F40

Fatal signal: 
Abort trap: 6

----- Backtrace -----
Backtrace unavailable
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.

This is a bug, please report it.
  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>
.

GDB closed TCP/IP connection (Socket 11)```

The problem does not arise with the original platform on (https://github.com/platformio/platform-raspberrypi.git). 

Do you have any ideas?

traxanos, 2023-10-05:
thats package is not working:

```
~/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: unrecognized option '--no-warn-rwx-segments'
~/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: use the --help option for usage information
```

but i don't think it's the toolchain. a gdb connection is established according to jlink. the firmware itself is functional.

maxgerhardt, 2023-10-05:
My apologies, you need both an older toolchain and older Arduino-Pico version that doesn't have that additional linker flag yet.

And I do think this toolchain change would help because it will then use a older GDB version instead of GDB13.

Please try:
```ini
platform_packages =
   toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-b/x86_64-apple-darwin14.arm-none-eabi-c7bab52.230216.tar.gz
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#73722b5c8731a6dd7a26b267bc9f920ce06c52bf
```

traxanos, 2023-10-05:
this combination works

```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages =
   toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-b/x86_64-apple-darwin14.arm-none-eabi-c7bab52.230216.tar.gz
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#73722b5c8731a6dd7a26b267bc9f920ce06c52bf
```

maxgerhardt, 2023-10-05:
Okay then sadly GDB13 is somehow not usable when the JLinkGDBServer(Exe) tool creates the GDBServer (CC @earlephilhower). Maybe there's a newer GDB13 version available now or we can recompile it with debug flags so that
```
----- Backtrace -----
Backtrace unavailable
---------------------
```
becomes an actual stack trace that can be filed at the bug-tracked of GDB.

traxanos, 2023-10-05:
I myself use the integrated (7.70) but had also manually started the jlink in a current version.

I could have just sworn that in a test with a 
toolchain 2.10xxx
framework: 3.4.1

but i can't get it to reproduce and get the same error again :(

maxgerhardt, 2023-10-05:
The package selection is "sticky", you need to explicitly remove the `platform_packages` and the old folders again. 
```
rm -rf ~/.platformio/packages/toolchain-rp2040-earlephilhower*
rm -rf ~/.platformio/packages/framework-arduinopico*
```

traxanos, 2023-10-05:
i had completely cleaned .platformio to be on the safe side. this always helps under windows too
somehow it all doesn't work out anymore. 

```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages =
   toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-b/x86_64-apple-darwin14.arm-none-eabi-c7bab52.230216.tar.gz
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#73722b5c8731a6dd7a26b267bc9f920ce06c52bf
```

```
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link EDU Mini V1 compiled Jul 31 2023 12:03:29
Hardware: V1.00
S/N: XXXXXXXX
Feature(s): FlashBP, GDB
Checking target voltage...
Target voltage: 3.29 V
Listening on TCP/IP port 2331
Connecting to target...
Connected to target
Waiting for GDB connection...
Reading symbols from ~/Projekte/OpenKNX/OAM-Dummy/.pio/build/develop_RP2040_DUALCORE_JLINK/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...
Connected to 127.0.0.1
Reading all registers
Read 2 bytes @ address 0x100322FC (Data = 0x4B02)
Read 4 bytes @ address 0x1002BB52 (Data = 0x00290020)
Reading 64 bytes @ address 0x20041EC0
Read 4 bytes @ address 0x1002BBAC (Data = 0x46C0BD70)
GDB closed TCP/IP connection (Socket 10)
```
I think I'll have to look over it again later.

GeminiServer, 2023-10-07:
I cleaned up the .platformio/ directory, and unfortunately, the same issue persists. I am still unable to debug.

<details open>
  <summary>Platformio.ini</summary>

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env]
;platform = raspberrypi
;platform = https://github.com/maxgerhardt/platform-raspberrypi.git

platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages =
   toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-b/x86_64-apple-darwin14.arm-none-eabi-c7bab52.230216.tar.gz
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#73722b5c8731a6dd7a26b267bc9f920ce06c52bf

framework = arduino
upload_speed = 115200
monitor_speed = 115200
debug_tool = jlink
upload_protocol = jlink

[env:rpipicow]
board = rpipicow  
```
</details>

<details>
    <summary>Console - Initial build after clean up</summary>

```console
***************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
***************************************************************************************************************************************************************************************************

Processing rpipicow (board: rpipicow; platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/maxgerhardt/platform-raspberrypi.git
git version 2.42.0
Cloning into '/Users/ecolak/.platformio/.cache/tmp/pkg-installing-7i3sbv13'...
remote: Enumerating objects: 122, done.
remote: Counting objects: 100% (122/122), done.
remote: Compressing objects: 100% (100/100), done.
remote: Total 122 (delta 73), reused 39 (delta 9), pack-reused 0
Receiving objects: 100% (122/122), 134.34 KiB | 2.86 MiB/s, done.
Resolving deltas: 100% (73/73), done.
Submodule 'builder/frameworks/arduino/mbed-core' (https://github.com/platformio/builder-framework-arduino-core-mbed.git) registered for path 'builder/frameworks/arduino/mbed-core'
Cloning into '/Users/ecolak/.platformio/.cache/tmp/pkg-installing-7i3sbv13/builder/frameworks/arduino/mbed-core'...
remote: Enumerating objects: 37, done.        
remote: Counting objects: 100% (37/37), done.        
remote: Compressing objects: 100% (27/27), done.        
remote: Total 37 (delta 10), reused 30 (delta 9), pack-reused 0        
Receiving objects: 100% (37/37), 11.70 KiB | 11.70 MiB/s, done.
Resolving deltas: 100% (10/10), done.
Submodule path 'builder/frameworks/arduino/mbed-core': checked out '6174f769bfc8610a397ed0c912cd059246e547cd'
Platform Manager: raspberrypi@1.10.0+sha.182d833 has been installed!
Tool Manager: Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-b/x86_64-apple-darwin14.arm-none-eabi-c7bab52.230216.tar.gz
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-rp2040-earlephilhower@5.100300.230216 has been installed!
[... 1033 lines cut ...]
Read register 'r2' (4 bytes) from hardware: 0x00200014
Read register 'r3' (4 bytes) from hardware: 0x00000000
Read register 'r4' (4 bytes) from hardware: 0x00000000
Read register 'r5' (4 bytes) from hardware: 0x011F0420
Read register 'r6' (4 bytes) from hardware: 0x00000018
Read register 'r7' (4 bytes) from hardware: 0x00000000
Read register 'r8' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r9' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r10' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r11' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r12' (4 bytes) from hardware: 0x1C800140
Read register 'sp' (4 bytes) from hardware: 0xE81E0420
Read register 'lr' (4 bytes) from hardware: 0x00000000
Read register 'pc' (4 bytes) from hardware: 0x9A340010
Read register 'xpsr' (4 bytes) from hardware: 0x00000041
Removing breakpoint @ address 0x1000349A, Size = 2
Removing breakpoint @ address 0x10004660, Size = 2
Read 4 bytes @ address 0x1000349A (Data = 0x024921E1)

Breakpoint 2, setup () at src/main.cpp:7
7	  Serial.begin(115200);
Reading 64 bytes @ address 0x20041EC0
GDB closed TCP/IP connection (Socket 11)
```
</details>

maxgerhardt, 2023-10-07:
So for one person it does with those toolchain + core version but for the other it doesn't? Hm. 

In any case, the original Raspberry platform also supports J-Link debugging, so why don't you try if this works (or is already broken there). I'd recommend again:
```
rm -rf ~/.platformio/platforms/raspberrypi
```
use [the standard example](https://github.com/platformio/platform-raspberrypi/tree/develop/examples/arduino-blink) with only 
```
upload_protocol = jlink
debug_tool = jlink
```
added. See if that is uploadable and debuggable.

GeminiServer, 2023-10-07:
Well, the interesting part is, that debugging with **original Raspberry** platform is **working** as expected. 

The upload/erase/etc. works in both cases.  
But, the pain here is, that the original raspi repo does actually not support the devices.

maxgerhardt, 2023-10-07:
Keep the same current project files (from arduino-blink) but use this `platformio.ini` instead
```ini
[env]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
debug_tool = jlink
upload_protocol = jlink

[env:pico]
board = pico
```
That is, it should use *my* platform but still the toolchain and framework packages from the original platform, which are ArduinoCore-mbed (not Arduino-Pico).

Does debugging still work? If yes, the fault is in the toolchain packages. If no, the fault is in the platform code.

GeminiServer, 2023-10-07:
Just did a quit setup with blink project:

**Conclusion:** Yes, the fault is in the toolchain packages.

**To Reproduce:**
1. Build and start Debug blink project with platform = raspberrypi
2. Build and start Debug blink with platform = https://github.com/maxgerhardt/platform-raspberrypi.git

After this steps, i was able to debug!

<details open>
  <summary>Platformio.ini</summary>

```ini

[env]
;platform = raspberrypi
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino

[env:pico]
board = pico

upload_protocol = jlink
debug_tool = jlink
```
</details>

[image]

GeminiServer, 2023-10-07:
**Conclusion:**  Yes, the fault lies within the toolchain packages.

**To reproduce:** Just did setup a blink project. 
<details open>
  <summary>Platformio.ini</summary>

```ini
[env]
;platform = raspberrypi
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino

[env:pico]
board = pico

upload_protocol = jlink
debug_tool = jlink
```
</details>

1. Build and start Debug project "Blink" using platform = raspberrypi 
> Debugging works!
2. Build and start Debug project "Blink" using platform = https://github.com/maxgerhardt/platform-raspberrypi.git
>Debugging works!

[image]

maxgerhardt, 2023-10-07:
I see. Then I only see 2 paths forward:
1. We downgrade the GDB version in the toolchain from 13.2 to something that works (but keep GCC at the same version)
2. We report this crash (with backtrace and reproduction instructions) to the GDB bugtacker. I've opened issue 31 as linked above for this.

Point 1 already worked for one person above, so I'm really not sure why it does not work on your machine. Can you do one final 
```
rm -rf ~/.platformio
```
and use the exact same `platformio.ini` as in https://github.com/maxgerhardt/platform-raspberrypi/issues/39#issuecomment-1751636869 and try this again?

GeminiServer, 2023-10-07:
## **Conclusion**
Debugging **works** with the **older GDB version** when using **board = pico**, **but not** when using **board = rpipicow**. 

If I use **board = rpipicow** with the following configuration, debugging is **not possible**. 
However, with **board = pico**, debugging runs as expected.

----------
_(Point 1 just worked for **one time**. He is not able to get it working again. --> https://github.com/maxgerhardt/platform-raspberrypi/issues/39#issuecomment-1748650884.)_
#### I tried it again and conducted further investigations. 
### **Here the steps to reproduce:**

1. The same "Blink" project like in: https://github.com/maxgerhardt/platform-raspberrypi/issues/39#issuecomment-1751636869
2. Using the one time working platform_pakages: https://github.com/maxgerhardt/platform-raspberrypi/issues/39#issuecomment-1748523186
3. Platform.ini -  Notice: I'm using "**rpi pico w**" !
<details> <summary>Platformio.ini</summary>

```ini
[env]
;platform = raspberrypi
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages =
   toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-b/x86_64-apple-darwin14.arm-none-eabi-c7bab52.230216.tar.gz
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#73722b5c8731a6dd7a26b267bc9f920ce06c52bf

framework = arduino

[env:pico]
board = pico
;board = rpipicow

upload_protocol = jlink
debug_tool = jlink
```
</details> 

7. Removed .platformio 
```console 
rm -rf ~/.platformio
```

8. Build and start Debug project "Blink" using platform_pakages
<details>
  <summary>Console - Initial build after clean up</summary>

```console
 *  Task wird ausgeführt: platformio run 

***********************************************************************************************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
***********************************************************************************************************************************************************************************************************************************************************************************************************

Processing pico (board: pico; platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/maxgerhardt/platform-raspberrypi.git
git version 2.42.0
Cloning into '/Users/ecolak/.platformio/.cache/tmp/pkg-installing-owthlgjs'...
remote: Enumerating objects: 122, done.
remote: Counting objects: 100% (122/122), done.
remote: Compressing objects: 100% (100/100), done.
remote: Total 122 (delta 73), reused 39 (delta 9), pack-reused 0
Receiving objects: 100% (122/122), 134.34 KiB | 2.80 MiB/s, done.
Resolving deltas: 100% (73/73), done.
Submodule 'builder/frameworks/arduino/mbed-core' (https://github.com/platformio/builder-framework-arduino-core-mbed.git) registered for path 'builder/frameworks/arduino/mbed-core'
Cloning into '/Users/ecolak/.platformio/.cache/tmp/pkg-installing-owthlgjs/builder/frameworks/arduino/mbed-core'...
remote: Enumerating objects: 37, done.        
remote: Counting objects: 100% (37/37), done.        
remote: Compressing objects: 100% (27/27), done.        
remote: Total 37 (delta 10), reused 30 (delta 9), pack-reused 0        
[... 1004 lines cut ...]
Read register 'r5' (4 bytes) from hardware: 0x011F0420
Read register 'r6' (4 bytes) from hardware: 0x00000018
Read register 'r7' (4 bytes) from hardware: 0x00000000
Read register 'r8' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r9' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r10' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r11' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r12' (4 bytes) from hardware: 0x1C800140
Read register 'sp' (4 bytes) from hardware: 0xE81E0420
Read register 'lr' (4 bytes) from hardware: 0x252B0010
Read register 'pc' (4 bytes) from hardware: 0x282B0010
Read register 'xpsr' (4 bytes) from hardware: 0x00000081
Read 4 bytes @ address 0x10002B28 (Data = 0x4B100122)
Removing breakpoint @ address 0x10000916, Size = 2
Removing breakpoint @ address 0x100029D2, Size = 2
Reading 64 bytes @ address 0x20041F00
2
WARNING: Failed to read memory @ address 0x60582020
````
</details>

**pipicow**
[image]
**pico**
[image]

maxgerhardt, 2023-10-07:
Well `pico` will use ArduinoCore-mbed with the "Pico" board (and the old toolchain) while `rpipicow` will use Arduino-Pico with "Pico W" board (and the new toolchain).

What happens when you set `board = rpipico`? This uses the "Pico" board with Arduino-Pico (and the new toolchain)

GeminiServer, 2023-10-07:
Using same settings, this time with board set to rpipico ```board = rpipico```.  Same behavior like the rpipicow.

[image]

maxgerhardt, 2023-10-07:
Can you try one last thing: To plantate the GDB from the working package into the non-working package.

1. Download the 9.3.1 toolchain-gccarmnoneeabi version for Mac from https://dl.registry.platformio.org/download/platformio/tool/toolchain-gccarmnoneeabi/1.90301.200702/toolchain-gccarmnoneeabi-darwin_x86_64-1.90301.200702.tar.gz
2. Delete all old `~/.platformio/packages/toolchain-rp2040-earlephilhower/bin/arm-none-eabi-gdb*` executables (there is gdb, gdb-add-index, gdb-add-index-py3, gdb-py)
3. Copy all `bin/arm-none-eabi-gdb*` binaries from the downloaded 9.3.1 package into ``~/.platformio/packages/toolchain-rp2040-earlephilhower/bin` along with all `.dylib` files.

GeminiServer, 2023-10-07:
Injected the working GDB files into the existing toolchain, and it seems to be working. 
Tested it with rpipicow, using the same previous setup.

--------

From: ```~/.platformio/packages/toolchain-rp2040-earlephilhower/bin```
**Removed:**
```
11M 12 Sep 02:12 arm-none-eabi-gdb
4,5K 12 Sep 02:12 arm-none-eabi-gdb-add-index 
```
**Added:** I think not all .dylib files are required. From: ``` 9.3.1 toolchain-gccarmnoneeabi```
```
7,1M  2 Jul  2020 arm-none-eabi-gdb
3,2K  2 Jul  2020 arm-none-eabi-gdb-add-index
3,2K  2 Jul  2020 arm-none-eabi-gdb-add-index-py3
7,5M  2 Jul  2020 arm-none-eabi-gdb-py3

202K  2 Jul  2020 libcrypt.2.dylib
175K  2 Jul  2020 libexpat.1.dylib
109K  2 Jul  2020 libgcc_s.1.dylib
477K  2 Jul  2020 libgmp.10.dylib
958K  2 Jul  2020 libiconv.2.dylib
57K  2 Jul  2020 libintl.8.dylib
159K  2 Jul  2020 liblzma.5.dylib
398K  2 Jul  2020 libmpfr.4.dylib
354K  2 Jul  2020 libncurses.6.dylib
2,5M  2 Jul  2020 libpython3.7m.dylib
2,4M  2 Jul  2020 libstdc++.6.dylib
```

[image]

GeminiServer, 2023-10-07:
Tested the injected GDB Env with my initial Setup! Debugger is working as expected.

maxgerhardt, 2023-10-07:
Okay, that's "working" then as a very temporary solution. We'll try and figure out what to do with toolchain downgrading or bugreporting in the linked pico-quick-toolchain issue. When we have a backtrace-enabled GDB, I would need someone to re-run the test though to get the full backtrace to be able to report. But that's for when it's ready.

GeminiServer, 2023-10-07:
Max, thank you for your support. This is a significant step forward, and we have identified the problem. It is currently running temporarily. Just let me know when I should rerun the test to obtain the backtrace.

earlephilhower, 2023-10-07:
Please look at https://github.com/earlephilhower/pico-quick-toolchain/issues/31#issuecomment-1751726139 and try with the rebuilt gdb executable.

maxgerhardt, 2025-08-30:
Closing due to stale issue and toolchain upgrade (GDB) and tool upgrades (tool-jlink) having happened in the meantime.

closed 2025-08-30 by maxgerhardt