maxgerhardt/platform-raspberrypi #42: support for PIO (pioasm)

vanbwodonk, 2023-10-13:
Hi @maxgerhardt, thank you for hard work! 

Just want to ask, is there any support for PIO (pioasm)?

tobigun, 2025-09-29:
@vanbwodonk even if your post is two years old, maybe it helps someone else.

It does not seem like there is any support for pioasm at the moment (@maxgerhardt: correct me if I am wrong).
So I just write my own extra_script for this.
[compile_pio.py](https://github.com/user-attachments/files/22596232/compile_pio.py)

If you add it to your platform.ini with
```
extra_scripts =
    compile_pio.py
```
it will compile the pio file to a header whenever you build. I am not sure why I did not use a PreAction but I think it did not execute pioasm before the code was compiled.

Note that you will have to manually download pioasm (contained in the pico-sdk) and change the path in the script accordingly.
Of cource also change the path to PIO_INPUT_FILE.

@maxgerhardt: would it be possible to include pioasm to packages\tool-rp2040tools? At least it would not be necessary anymore to download pioasm manually.

vanbwodonk, 2025-10-02:
@tobigun @maxgerhardt 
I've been using this gist tutorial for a while now  [link](https://gist.github.com/hexeguitar/f4533bc697c956ac1245b6843e2ef438).
It uses Earle's toolchain, as suggested by @maxgerhardt. I'd be glad if this becomes officially supported.