maxgerhardt/platform-raspberrypi #43: FreeRTOS support

mchacher, 2023-10-16:
Hi, while browsing the web I have been able to compile a project (not tested yet) with FreeRTOS. I would like to leverage SMP / dual core capability of the RP2040. 

- Is it fully supported with current implementation with capability to assign tasks to the 2 cores and enable Queue communication between the tasks?
- Can you confirm that the "official" platformio is not supporting FreeRTOS? 
- To use FreeRTOS I have changed platformio.ini file like this. Although I have understood that Arduino IDE is supporting FreeRTOS. Can you clarify a bit the panorama and confirm that wwhat I did is correct? 

Thanks a lot.

```
; with FreeRTOS support
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi
; platform = raspberrypi
board = pico
framework = arduino
board_build.core = earlephilhower
```

vs.

```
; without FreeRTOS support
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
```

mchacher, 2023-10-16:
Adding to my previous post, I have a compilation issue on a large project that I don't understand.
<command-line>: error: expected identifier before numeric constant
```
/Users/mchacher/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h:155:7: note: in expansion of macro 'RP2040'
  155 | class RP2040;
```
I am working with platformio / vscode on MacOS. Looking at the file RP2040Support.h it looks like this file is old, and not in sync https://github.com/earlephilhower/arduino-pico.

I wonder whether my setup is correct.

Thanks for your support.

AgainPsychoX, 2024-03-21:
Is there maybe good write up about differences between Arduino-Pico (aka `earlephilhower` version) and PlatformIO supported one, the ArduinoCore-mbed? Or anything you suggest I should read/look up?

closed 2024-09-15 by maxgerhardt

znmeb, 2024-09-15:
Longer term, Arduino is phasing out MBED in favor of Zephyr, not FreeRTOS.

https://blog.arduino.cc/2024/07/24/the-end-of-mbed-marks-a-new-beginning-for-arduino/
https://blog.adafruit.com/2024/08/01/arduino-to-switch-from-arm-mbed-to-zephyr-real-time-operating-system-rtos/

Zephyr is on my list of systems to learn because I have a number of embedded devices that they support, including a few FPGA dev boards.