maxgerhardt/platform-raspberrypi #47: Error during installation

fabianoriccardi, 2023-11-18:
During the installation of this core, I get the following error which stops the installation. My configuration:

Platformio.ini:
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
board = pico
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
```

Log of Output console:

```
......

Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32u5xx_hal_driver': checked out '2e1d4cdb386e33391cb261dfff4fefa92e4aa35a'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32wbxx_hal_driver': checked out '2c5f06638be516c1b772f768456ba637f077bac8'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/ti': checked out '143ed6cc20a7615d042b03b21e070197d473e6e5'
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long

error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long

From https://github.com/ARM-software/CMSIS_5
 * branch                20285262657d1b482d132d20d755c8c330d55c1f -> FETCH_HEAD

Submodule path 'pico-sdk/lib/tinyusb/lib/CMSIS_5': checked out '20285262657d1b482d132d20d755c8c330d55c1f'
Submodule path 'pico-sdk/lib/tinyusb/lib/FreeRTOS-Kernel': checked out '2a604f4a2818b8354b5e1a39e388eb5e16cfbc1f'
From https://github.com/lwip-tcpip/lwip
 * branch              159e31b689577dbf69cf0683bbaffbd71fa5ee10 -> FETCH_HEAD

Submodule path 'pico-sdk/lib/tinyusb/lib/lwip': checked out '159e31b689577dbf69cf0683bbaffbd71fa5ee10'
Submodule path 'pico-sdk/lib/tinyusb/lib/sct_neopixel': checked out 'e73e04ca63495672d955f9268e003cffe168fcd8'
Submodule path 'pico-sdk/lib/tinyusb/tools/uf2': checked out '19615407727073e36d81bf239c52108ba92e7660'
Submodule 'hidapi' (https://github.com/signal11/hidapi) registered for path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi'

Cloning into 'C:/Users/xxx/.platformio/.cache/tmp/pkg-installing-4sdl6qpl/pico-sdk/lib/tinyusb/tools/uf2/hidapi'...

Submodule path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
fatal: Unable to checkout '17761f5cf9dbbf2dcf665b7c04934188add20082' in submodule path 'pico-sdk/lib/tinyusb/hw/mcu/wch/ch32v307'

fatal: Failed to recurse into submodule path 'pico-sdk/lib/tinyusb'

From https://github.com/earlephilhower/bearssl-esp8266
 * branch            f294aa022f82833941ffdf3b0f306c2072e3a26c -> FETCH_HEAD

Submodule path 'tools/libbearssl/bearssl': checked out 'f294aa022f82833941ffdf3b0f306c2072e3a26c'
Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
fatal: Failed to recurse into submodule path 'pico-sdk'

VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\xxx\\.platformio\\.cache\\tmp\\pkg-installing-4sdl6qpl']
```

Am I missing any dependency?  Does `platformio.ini` contain an error?

fabianoriccardi, 2023-11-19:
I have tried different commits down to 79e4f615a10d29d87edc9e69bbf28b9f4afc41cc (24 May 2023), but all fails at the same step.

nWestie, 2024-02-23:
See my comment on [#52](https://github.com/maxgerhardt/platform-raspberrypi/issues/52#issuecomment-1961401018) - it looks like you didn't get long-paths enabled successfully.

maxgerhardt, 2024-04-29:
>Filename too long

Yep, going to attribute this one to NTFS long paths. Note was added in main README, too.

closed 2024-04-29 by maxgerhardt