maxgerhardt/platform-raspberrypi #53: Feature Request / Info: earlephilhower Updated toolchains -> native Apple ARM support

Jason2866, 2024-01-26:
@maxgerhardt maybe updating to latest release of the toolchains and using the native Apple ARM?
https://github.com/earlephilhower/pico-quick-toolchain/releases/tag/2.2.0

maxgerhardt, 2024-01-26:
Mhm, I've seen that release too, will check out. Should be easy to integrate since we already have the dictionary from sys type to download URL.

Jason2866, 2024-01-26:
I saw your solution for dl the system matching toolchain from github repo. Nice solution when not wanting to upload and maintain in Platformio registry!

maxgerhardt, 2024-01-26:
Let's see how eef7c2908b27788fbcf6b343a71c4a26b5727f47 goes through the CI...

maxgerhardt, 2024-01-26:
Yes, CI goes through, make you can update your platform version and retest on native ARM Mac?

Jason2866, 2024-01-27:
Does work :-) 
Using the git dev platform, cloning tinyusb needs a looong time. Considering to add as separate package?
```
 *  Task wird ausgeführt: platformio run --environment pico_earle 

Processing pico_earle (board: pico; platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino)
-----------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.11.0+sha.da81492) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30603.0+sha.adb23c1 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 51 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/pico_earle/FrameworkArduinoBootloader/boot2_w25q080_2_padded_checksum.S.o
Compiling .pio/build/pico_earle/src/main.cpp.o
Generating linkerscript /Volumes/T7-Mac/platform-raspberrypi/examples/arduino-blink/.pio/build/pico_earle/memmap_default.ld
Compiling .pio/build/pico_earle/FrameworkArduino/BluetoothDebug.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/Bootsel.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/CoreMutex.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/FS.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/PIOProgram.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/RP2040Support.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/RP2040USB.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/SerialPIO.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/SerialUART.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/SerialUSB.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/StackThunk.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/_freertos.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/api/Common.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/api/IPAddress.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/api/PluggableUSB.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/api/Print.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/api/Stream.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/api/String.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/cyw43_wrappers.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/delay.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/libb64/cdecode.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/libb64/cencode.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/lock.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/lwip_wrap.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/main.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/malloc-lock.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/posix.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/sdkoverride/att_db.c.o
Compiling .pio/build/pico_earle/FrameworkArduino/sdkoverride/btstack_flash_bank.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/sdkoverride/hids_device.c.o
Compiling .pio/build/pico_earle/FrameworkArduino/sdkoverride/pico_bootsel_via_double_reset.c.o
Compiling .pio/build/pico_earle/FrameworkArduino/stdlib_noniso.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/wiring_analog.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/wiring_digital.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/wiring_private.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/wiring_pulse.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/wiring_shift.cpp.o
Archiving .pio/build/pico_earle/libFrameworkArduino.a
Indexing .pio/build/pico_earle/libFrameworkArduino.a
Linking .pio/build/pico_earle/firmware.elf
Generating UF2 image
elf2uf2 ".pio/build/pico_earle/firmware.elf" ".pio/build/pico_earle/firmware.uf2"
Retrieving maximum program size .pio/build/pico_earle/firmware.elf
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
Checking size .pio/build/pico_earle/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.8% (used 10040 bytes from 262144 bytes)
Flash: [          ]   2.8% (used 57960 bytes from 2093056 bytes)
Building .pio/build/pico_earle/firmware.bin
Building .pio/build/pico_earle/firmware.bin.signed
======================================= [SUCCESS] Took 9.10 seconds =======================================

Environment    Status    Duration
-------------  --------  ------------
pico_earle     SUCCESS   00:00:09.104
======================================= 1 succeeded in 00:00:09.104 =======================================
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. 
 ```

Jason2866, 2024-01-27:
Stop something is wrong!
This looks fine
```
Tool Manager: Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.2.0/x86_64-apple-darwin20.4.arm-none-eabi-d04e724.240125.tar.gz
```
but
package.json
```
{
   "description": "arm-none-eabi-gcc",
   "name": "toolchain-rp2040-earlephilhower",
   "system": [ "darwin_x86_64" ],
   "url": "https://github.com/earlephilhower/pico-quick-toolchain",
   "version": "5.120300.240125"
}
```

maxgerhardt, 2024-01-27:
Seems to me like PlatformIO is detecting your system to be regular x86_64 instead of native ARM. For this to work, you need to be running VSCode, Python and PlatformIO etc. all in native ARM. 

PlatformIO just queries what Python is returning as `platform.machine()`, `platform.system()`, etc.

https://github.com/platformio/platformio-core/blob/develop/platformio/util.py#L138-L148

You may need to `rm -rf ~/.platformio` and check that your VSCode binary is native ARM.

Jason2866, 2024-01-27:
@maxgerhardt Weird since calling https://github.com/earlephilhower/arduino-pico/pull/1959#issuecomment-1912552863 from VSC shows all native ARM

maxgerhardt, 2024-01-27:
But does this happen with the Python environment of PlatformIO? You would need to first 
```
cd ~/,platformio/penv/bin
source activate
```

Jason2866, 2024-01-27:
Grrr. The update of platformio core (just happend this morning) installs a `darwin_x86_64` pyenv.
This is a bug in platformio core in latest version.

Jason2866, 2024-01-27:
Swapped pyenv to native ARM one. Now the native ARM toolchain is loaded. And compile does hang endless

```
 *  Task wird ausgeführt: platformio run --environment pico_earle 

Processing pico_earle (board: pico; platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino)
-----------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.11.0+sha.da81492) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30603.0+sha.adb23c1 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 51 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/pico_earle/FrameworkArduinoBootloader/boot2_w25q080_2_padded_checksum.S.o
Compiling .pio/build/pico_earle/src/main.cpp.o
Generating linkerscript /Volumes/T7-Mac/platform-raspberrypi/examples/arduino-blink/.pio/build/pico_earle/memmap_default.ld
Compiling .pio/build/pico_earle/FrameworkArduino/BluetoothDebug.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/Bootsel.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/CoreMutex.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/FS.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/PIOProgram.cpp.o
Compiling .pio/build/pico_earle/FrameworkArduino/RP2040Support.cpp.o
```

maxgerhardt, 2024-01-27:
That's.. bad. When you manually execute the compiler on the shell, does it at least start up, or fail because of some missing dylibs maybe?
```
~/.platformio/packages/toolchain-rp2040-earlephilhower/bin/arm-none-eabi-gcc --version
```

Jason2866, 2024-01-27:
Dead at start :-(
```
hans@MacBook-Air-von-Johann arduino-blink % ~/.platformio/packages/toolchain-rp2040-earlephilhower/bin/arm-none-eabi-gcc --version
```
nothing happens

Btw. opened https://github.com/platformio/platformio-vscode-ide/issues/3949

Jason2866, 2024-01-27:
Trying to start any of the binaries ends in killed process. I had this behaviour too when i compiled the xtensa espressif32 toolchains myself. For a reason (i dont know) the strip command at the end of the build process corrupts the binary

Jason2866, 2024-01-27:
I would revert the commit. Native ARM toolchains WILL get installed. It is a major bug currently of platformio-vscode-ide introduced with v3.3.0. This bug needs to be fixed soon from Platfromio since it breaks ALL use of native Apple ARM tools and toolchains. This worked well before this update.

maxgerhardt, 2024-01-27:
Reverted back to darwin_x86_64 for Mac ARM, which should work fine as previous, in a025e1a7666f46ecc265262257485a5ede2dbec2.

Jason2866, 2024-01-27:
@maxgerhardt  This is not working either since the package.json has no entry for ARM

```
{
   "description": "arm-none-eabi-gcc",
   "name": "toolchain-rp2040-earlephilhower",
   "system": [ "darwin_x86_64" ],
   "url": "https://github.com/earlephilhower/pico-quick-toolchain",
   "version": "5.120300.240125"
}
```

maxgerhardt, 2024-01-27:
Right, then it must use the older 2.1.0-a toolchain. 55d5d52cf1c12d295dc7cdef6f593af0f14d1e0e

Jason2866, 2024-01-27:
@maxgerhardt It was indeed the `strip` command at the end of the build process.
earlephilhower re-built without `strip` and the toolchain is now working :-)

maxgerhardt, 2024-01-27:
Seems like only the URL to the Apple ARM toolchain changed, all others are still the same. Adapted to in 1f63e598ee639fc98cf133fbe2744415d01e60e1.

Jason2866, 2024-01-28:
Verified, working.

closed 2024-01-28 by Jason2866