maxgerhardt/platform-raspberrypi #55: SPI1 on SD

BaTaTaAdb, 2024-02-10:
When using the SD library, the compiler uses the arduino SD library, which doesn't have the overflow version `begin(uint8_t csPin, uint32_t cfg = SPI_HALF_SPEED, HardwareSPI &spi = SPI)` which I need in order to have it on SPI1 (correct me if I'm wrong).

This is the part of my code that is giving me the error `if (!SD.begin(CS_SPI, SPI1))`.
How do I override the arduino library with the custom board library to be able to use SPI1 on my board? SPI1 is necessary in my case.

platformio.ini:
`[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
lib_deps = 
	olikraus/U8g2@^2.35.9
	adafruit/Adafruit Unified Sensor@^1.1.14
	adafruit/Adafruit BNO055@^1.6.3
	adafruit/Adafruit BME280 Library@^2.2.4
	adafruit/Adafruit GPS Library@^1.7.4
`
If you need more info, let me know.

maxgerhardt, 2024-04-29:
What's the "Overflow version"?

Arduino-Pico's SD library does allow you to use whatever SPI port you want. If it uses a different SD library, use [`lib_ignore`](https://docs.platformio.org/en/latest/projectconf/sections/env/options/library/lib_ignore.html) against the undesired one.

https://github.com/earlephilhower/arduino-pico/blob/master/libraries/SD/src/SD.h#L39-L42

BaTaTaAdb, 2024-04-29:
This question was made almost 2 months ago. The problem was that platformio wasn't recognizing any of pico libraries. I'm not quite sure what fixed it, but I remember that at some point I just created a customSD library that was a copy of the pico SD library.

closed 2024-04-29 by maxgerhardt