maxgerhardt/platform-raspberrypi #56: Filesystem-size >= 2MB bricks COM port

L0rdL4nx, 2024-02-25:
I'm using a custom RP2040 board with 16MB flash and the `generic` preset in Platform Io.
Once I define a filesystem size higher or equal to 2MB and upload, my board wont boot as a COM device anymore. It defaults to UF2 boot, even without pressing BOOTSEL once.
Which wont be a problem if i could use uf2conv for Serial Monitor, which isnt available.
When uploading in Arduino IDE with a high filesystem size, e.g. 8MB out of 16 available, it works as espected.

Does anyone know what to do in this szenario? I already tried reinstalling everything and updating all drivers.

maxgerhardt, 2024-04-29:
Full PlatformIO for reference?

L0rdL4nx, 2024-05-03:
```ini
[env:calibration]
build_src_filter = +<calibration.cpp>
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = generic
framework = arduino
board_build.core = earlephilhower
; define a filesystem size from available maximum size (my custom board has 16MB, a standard Pi Pico has 2MB)
; expression can involve "b", "k", "m" (bytes/kilobytes/megabytes) and floating point numbers
board_build.filesystem_size = 8m
board_build.f_cpu = 133000000L
lib_deps = bblanchon/ArduinoJson@^7.0.2
```

Heres the .ini snippet.
The platform io project is otherwise pretty much empty, my file (calibration.cpp) contains a simple Serial-Testscript.

Platform IO version is `3.3.3`
RP2040 core version is `1.11.0+sha.60d6ae8`

L0rdL4nx, 2024-06-30:
I still haven't solved it, migrated to the official Arduino IDE for now.
Was above information what you wanted or are there any other logs that could help with finding the problem?

L0rdL4nx, 2024-09-02:
I did a little oopsie on my part, sorry for that. 
Silly me thought the W25Q16JVxx was the 16MB version, not the 16Mbit = 2MByte flash chip. How's flash size greater than 2MB going to work with that, huh?
So no problem from software side - the biggest problem was infront of the PC.

closed 2024-09-02 by L0rdL4nx