maxgerhardt/platform-raspberrypi #63: _exit.dbgasm called when debugging with VSCode, Pico becomes non-responsive

mytechnotalent, 2024-04-21:
When attempting to debug this code, the `exit.dbgasm` is being called and causing the Pico to be unresponsive.

CODE
```#include <Arduino.h>
#include "pico/stdlib.h"
#include "hardware/uart.h"

const int UART_TX_PIN = 12; // GPIO0 is TX for UART0
const int UART_RX_PIN = 13; // GPIO1 is RX for UART0
const int BAUD_RATE = 115200;

void setup() {
    // Initialize UART0
  uart_init(uart0, BAUD_RATE);
  
  // Set TX and RX pins for UART0
  gpio_set_function(UART_TX_PIN, GPIO_FUNC_UART);
  gpio_set_function(UART_RX_PIN, GPIO_FUNC_UART);
  
  // Enable UART0 receiver and transmitter
  uart_set_hw_flow(uart0, false, false);
  uart_set_format(uart0, 8, 1, UART_PARITY_NONE);
  Serial.begin(115200);
}

void loop() {
  volatile uint8_t num = 42;
  uart_puts(uart0, "AT+SEND=0,1,K\r\n");
  sleep_ms(10);
}
```
CONFIG:
```
[env:rpipico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipico
framework = arduino
board_build.core = earlephilhower
debug_tool = cmsis-dap
```

maxgerhardt, 2024-04-29:
Does it also crash and become unresponsive when you load the same firmwarecode in the regular Arduino IDE with the Arduino-Pico core installed? If yes, better open an issue in [Arduino-Pico](https://github.com/earlephilhower/arduino-pico/issues) itself.

mytechnotalent, 2024-04-29:
HI @maxgerhardt first thank you for getting back. We use your work on just about everything our DEFCON local chapter codes and really appreciate your work. I have not tested on the Arduino IDE as I really prefer PlatformIO however will test for you and report back.

mytechnotalent, 2024-04-29:
I was unable to reproduce in Arduino IDE.

maxgerhardt, 2025-09-06:
I'll mark this as resolved for now, tested locally again with your code and I don't get a call into exit(). 

[image]

Of course, the original code is heavily mixing direct Pico-SDK calls with regular Arduino code (`Serial`) so maybe that was also the cause of the problem.

closed 2025-09-06 by maxgerhardt