maxgerhardt/platform-raspberrypi #68: Upload task times out for Raspberry Pi Pico

mustafaoral, 2024-08-08:
I have recently upgraded to Windows 11 (fresh install). I wanted to get back to working on my project, which had been sitting idle for a few months, but I'm having issues.

When I ran the Upload task, it failed with the long string of dots, usually something I see when I forget to close the Serial Monitor in TyCommander. I opened up the CLI and used the command `pio run -v -t upload`. It showed the following:

```
<lambda>(["upload"], [".pio\build\pico\firmware.elf"])
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
BeforeUpload(["upload"], [".pio\build\pico\firmware.elf"])
TimeoutError: Could not automatically find serial port for the `Raspberry Pi Pico` board based on the declared HWIDs=['2E8A:00C0']
TimeoutError: Could not automatically find serial port based on the known UART bridges
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
```

Oddly, I only get this output if I specify the port using VS Code UI. If I specify the port in platformio.ini, then I get the following:

```
<lambda>(["upload"], [".pio\build\pico\firmware.elf"])
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
BeforeUpload(["upload"], [".pio\build\pico\firmware.elf"])
Using manually specified: COM3
Forcing reset using 1200bps open/close on port COM3
"C:\Users\moss\.platformio\packages\tool-rp2040tools\rp2040load" -v -D .pio\build\pico\firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
```

The device is definitely on COM3. When I run the devices task, then I just get the following:
```
COM3
----
Hardware ID: USB VID:PID=2E8A:00C0 SER=083061E62D7332E3
Description: USB Serial Device (COM3)
```

Here's the content of my platform.io file:
```ini
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
upload_port = COM3 #left this here as an example. whether this line exists or not doesn't make a difference
lib_deps =
    olikraus/U8g2@^2.35.19
```

Output of dependencies - list task
```
Resolving pico dependencies...
Platform raspberrypi @ 1.14.0 (required: raspberrypi)
├── framework-arduino-mbed @ 4.1.5 (required: platformio/framework-arduino-mbed @ ~4.1.5)
├── tool-openocd-raspberrypi @ 2.1100.0 (required: platformio/tool-openocd-raspberrypi @ ~2.1100.0)
├── tool-rp2040tools @ 1.0.2 (required: platformio/tool-rp2040tools @ ~1.0.2)
└── toolchain-gccarmnoneeabi @ 1.90201.191206 (required: platformio/toolchain-gccarmnoneeabi @ ~1.90201.0)

Libraries
└── U8g2 @ 2.35.19 (required: olikraus/U8g2 @ ^2.35.19)
```

VS Code:
```
Version: 1.92.1 (user setup)
Commit: eaa41d57266683296de7d118f574d0c2652e1fc4
Date: 2024-08-07T20:16:39.455Z
Electron: 30.1.2
ElectronBuildId: 9870757
Chromium: 124.0.6367.243
Node.js: 20.14.0
V8: 12.4.254.20-electron.0
OS: Windows_NT x64 10.0.22631
```

maxgerhardt, 2024-08-09:
```
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
```
This error will be resolved if you use https://zadig.akeo.ie/, plug the Pico in with BOOTSEL pressed, then use WinUSB or libusb drivers for `RP2 Boot2 (Interface 1)`, as referenced in https://github.com/platformio/platform-raspberrypi/issues/2#issuecomment-828586398

mustafaoral, 2024-08-10:
That worked, thank you.

In your linked comment, you said that you though it possible that the upload may have worked for people who had Arduino IDE installed. On my old drive with Windows 10, I first used Arduino IDE before switching to PlatformIO so perhaps that's how it "just worked" when I did the switch.

On a side note, I realize now that I may have opened this issue at the wrong repository. Apologies for that.

closed 2024-09-15 by maxgerhardt