maxgerhardt/platform-raspberrypi #73: [SUGGESTION] Join and move this forked platform-raspberrypi repo to the new pioarduino community orginazation on GitHub?

Hedda, 2024-09-06:
@maxgerhardt I would just like to informally raise the suggestion of maybe you considering the idea of possibly joining the pioarduino community and perhaps moving this forked platform-raspberrypi repo to the new "pioarduino" community's organization on GitHub?

* https://github.com/pioarduino

If you have not yet of the new "pioarduino" community and its organization on GitHub and want the full backstory this check out this:

* https://github.com/espressif/arduino-esp32/discussions/10039

cnx-software also wrote a blog post about the new pioarduino community and their platform-espressif32 fork which sums it up, see:

* https://www.cnx-software.com/2024/08/27/platform-espressif32-fork-platformio-arduino-esp32-c6-esp32-c5-esp32-h2-esp32-p4/

Anyway, sounds like you are in a similar situation due to conflict between PlatformIO and Raspberry Pi Foundation / Raspberry Pi Ltd.

* https://github.com/platformio/platform-raspberrypi/pull/36

That is, PlatformIO is now also refusing community contributions to improve its platform-raspberrypi library, so it has stalled now too:

  * https://github.com/platformio/platform-raspberrypi/pull/70

Back-story summery on the pioarduino community as I understand that "pioarduino" (were pioarduino stands for “people initiated optimized arduino” and today includes Tasmota and ESPEasy developers) is a new open-source community initially formed by [Jason2866](https://github.com/Jason2866) a few weeks ago after PlatformIO's upstream development [platform-espressif32 (Arduino core for ESP32)](https://github.com/pioarduino/platform-espressif32) stalled because of some ongoing conflict between PlatformIO and Espressif which have practically stopped almost all development in upstream, (which I understand is a similar situation as platform-raspberrypi?), so when it  became clear the PlatformIO was no longer going to maintained or accepting contributions to upstream, [Jason2866](https://github.com/Jason2866) started a new fork of the platform-espressif32 reposotory and put that in a new community organization on GitHub so others can contribute or even join and help maintain it.

* https://github.com/pioarduino/platform-espressif32

pioarduino Discord Server for coordinating and early support:

* https://discord.gg/Nutz9crnZr

Docu for using pioarduino:

* https://github.com/pioarduino/docs

_Originally posted by [Jason2866](https://github.com/Jason2866) in https://github.com/espressif/arduino-esp32/discussions/10039_

> _**Since it is now clear that there will be no official support for Arduino core 3.0.x from Platformio team, ([Issue No. 1225](https://github.com/platformio/platform-espressif32/issues/1225) in repo https://github.com/platformio/platform-espressif32/issues) i decided to fork the needed repo(s) and build a community version to support core 3.0.x (starting with core 3.0.3)**_

So pioarduino's fork of platform-espressif32 accept communiy patches and is today way ahead of upstream, similar to your fork. And If an ESP32 user want to want to use Espressif Arduino 3.0.x and IDF 5.x.x or newer with ESP32 SoCs like ESP32-C6 and ESP32-H2 via platformio then the best choice is probably to use that fork, and contribute back to that fork instead of trying to get upstream to accept patches.

PS: For additional information on the platform-espressif32 situation please refer to these GitHub links:

* https://github.com/espressif/arduino-esp32/discussions/10039
* https://github.com/platformio/platform-espressif32/issues/1225
* https://github.com/espressif/arduino-esp32/pull/8606

Jason2866, 2024-09-06:
Currently i see no real benefit for `platform-raspberrypi` to move to pioarduino org.
The situation will change if using the Platformio Registry gets problematic or is not possible at all. pioarduino has already a full working PoC for building espressif32 Arduino and IDF projects **without** any use of Platformio registry.
If this is a (future) goal of `platform-raspberrypi` too, there are synergy effects how to achieve and optimise this.

Hedda, 2024-09-06:
> The situation will change if using the Platformio Registry gets problematic or is not possible at all. pioarduino has already a full working PoC for building espressif32 Arduino and IDF projects [...]

FYI, I think many RP2040/RP2350 (RPi Pico) users who don't use Platformio are instead using arduino-pico by @earlephilhower

* https://github.com/earlephilhower/arduino-pico

* https://arduino-pico.readthedocs.io/en/latest/

Hedda, 2024-09-23:
> Currently i see no real benefit for `platform-raspberrypi` to move to pioarduino org.
> The situation will change if using the Platformio Registry gets problematic or is not possible at all. [...]

To clearify, Platformio Registry is now problematic for those who want to use the new Raspberry Pi RP2350 and board for it:

* https://github.com/platformio/platform-raspberrypi/issues/67

For reference; PlatformIO's CEO have more or less made it clear that they will not allow their upstream platform-raspberrypi to keep being developed in their repository/repositories as long as Raspberry Pi Foundation or Raspberry Pi Ltd is not officially paying PlatformIO to maintain it there, and infer that the decision to discontinue support for upstream PlatformIO platforms are entirly up to the companies paying PlatformIO to maintain each platform. See:

* https://github.com/platformio/platform-raspberrypi/pull/36#issuecomment-1501724359

PlatformIO has not taken community contributions to improve its platform-raspberrypi platform library in their upstream repository/repositories for 2-years now, niether are they pushing any Raspberry Pi specific imporvements, all of which means that upstream development of PlatformIO's platform library for Raspberry Pi Pico(s) have stalled or put on hold for now. Because of this users feel forced to move to forks or alternative Arduino-Pico core without PlatformIO in order to use the new RP2350 MCU/SoC boards, including the Raspberry Pi Pico 2.

znmeb, 2024-09-23:
I'm not sure which alternative I'd use moving forward with my main project, https://github.com/AlgoCompSynth/CLAMS. The initial scope was limited to the RP2040 and a mixed C / assembly implementation using the Pico SDK, but a few months ago I decided to go for portability over raw performance.

Then the surprise announcement of the RP2350 happened, giving me another opportunity to change the scope. :-) So I ordered a few boards and spent a couple of weeks upgrading my development environment for them, then decided to investigate alternatives for portability.

As far as I know, all the boards I want to target for porting and system libraries I need are supported by `arduino-cli`, so it's not clear I need PlatformIO. But `cforth` uses it, and that pretty much reduces the CLAMS project to the part that interests me, Forth application programming. So I can go either way, but the existence of a community support mechanism for PlatformIO is a strong plus for staying with `cforth` and PlatformIO.

Jason2866, 2025-08-03:
@nicksterious  Platformio is doing business. It is there decision what the do or not. Since it is licensed with Apache 2 forks are possible.