maxgerhardt/platform-raspberrypi #75: Compiling for Pico2 fail

sstaub, 2024-09-07:
I try to compile a simple blinky example but it fails.
Here the output with error:

```
 *  Executing task: platformio run 

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino; board: rpipico2)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipico2.html
PLATFORM: Raspberry Pi RP2040 (1.14.0+sha.f519f65) > Pico 2
HARDWARE: RP2350 150MHz, 512KB RAM, 4MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40001.0+sha.6a5f98c 
 - tool-picotool-rp2040-earlephilhower @ 5.120300.240827 (12.3.0) 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 4.00MB
Sketch size: 3.50MB
Filesystem size: 0.50MB
Maximium Sketch size: 3665920 EEPROM start: 0x103ff000 Filesystem start: 0x1037f000 Filesystem end: 0x103ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 61 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/pico/src/main.cpp.o
Compiling .pio/build/pico/FrameworkArduino/Bootsel.cpp.o
Compiling .pio/build/pico/FrameworkArduino/CoreMutex.cpp.o
Compiling .pio/build/pico/FrameworkArduino/FS.cpp.o
Compiling .pio/build/pico/FrameworkArduino/PIOProgram.cpp.o
Compiling .pio/build/pico/FrameworkArduino/RP2040Support.cpp.o
Compiling .pio/build/pico/FrameworkArduino/RP2040USB.cpp.o
Compiling .pio/build/pico/FrameworkArduino/SerialPIO.cpp.o
Compiling .pio/build/pico/FrameworkArduino/SerialUART.cpp.o
Compiling .pio/build/pico/FrameworkArduino/SerialUSB.cpp.o
Compiling .pio/build/pico/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/pico/FrameworkArduino/WMath.cpp.o
In file included from src/main.cpp:1:
/Users/sstaub/.platformio/packages/framework-arduinopico/cores/rp2040/Arduino.h:30:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   30 | #include <pins_arduino.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/pico/src/main.cpp.o] Error 1
In file included from /Users/sstaub/.platformio/packages/framework-arduinopico/cores/rp2040/Bootsel.cpp:6:
/Users/sstaub/.platformio/packages/framework-arduinopico/cores/rp2040/Arduino.h:30:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

   30 | #include <pins_arduino.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/pico/FrameworkArduino/Bootsel.cpp.o] Error 1
In file included from /Users/sstaub/.platformio/packages/framework-arduinopico/cores/rp2040/CoreMutex.cpp:24:
[... 130 lines cut ...]
*
**********************************************************************

   30 | #include <pins_arduino.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/pico/FrameworkArduino/Tone.cpp.o] Error 1
=================================================== [FAILED] Took 1.08 seconds ===================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

here the .ini file
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
board = rpipico2
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#master
build_flags = -Wall
```

maxgerhardt, 2024-09-08:
Does the file `/Users/sstaub/.platformio/packages/framework-arduinopico/variants/rpipico2/pins_arduino.h` exist on your harddrive?

sstaub, 2024-09-08:
The complete folder rpipico2 is missing

maxgerhardt, 2024-09-08:
`rm -rf /Users/sstaub/.platformio/packages/framework-arduinopico/` and press the build button again.

sstaub, 2024-09-08:
Fails also
```
 *  Executing task: platformio run 

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino; board: rpipico2)
----------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/earlephilhower/arduino-pico.git#master
git version 2.39.2
Cloning into '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571'...
remote: Enumerating objects: 1584, done.
remote: Counting objects: 100% (1584/1584), done.
remote: Compressing objects: 100% (1203/1203), done.
remote: Total 1584 (delta 388), reused 1050 (delta 141), pack-reused 0 (from 0)
Receiving objects: 100% (1584/1584), 9.80 MiB | 7.49 MiB/s, done.
Resolving deltas: 100% (388/388), done.
Submodule 'ArduinoCore-API' (https://github.com/earlephilhower/ArduinoCore-API.git) registered for path 'ArduinoCore-API'
Submodule 'cores/rp2040/tlsf' (https://github.com/earlephilhower/tlsf.git) registered for path 'lib/tlsf'
Submodule 'libraries/Adafruit_TinyUSB_Arduino' (https://github.com/adafruit/Adafruit_TinyUSB_Arduino.git) registered for path 'libraries/Adafruit_TinyUSB_Arduino'
Submodule 'libraries/AsyncUDP' (https://github.com/earlephilhower/AsyncUDP.git) registered for path 'libraries/AsyncUDP'
Submodule 'libraries/SdFat' (https://github.com/earlephilhower/ESP8266SdFat.git) registered for path 'libraries/ESP8266SdFat'
Submodule 'libraries/FatFS/lib/SPIFTL' (https://github.com/earlephilhower/SPIFTL.git) registered for path 'libraries/FatFS/lib/SPIFTL'
Submodule 'libraries/FreeRTOS/lib/FreeRTOS-Kernel' (https://github.com/earlephilhower/FreeRTOS-Kernel.git) registered for path 'libraries/FreeRTOS/lib/FreeRTOS-Kernel'
Submodule 'libraries/Joystick' (https://github.com/earlephilhower/Joystick.git) registered for path 'libraries/HID_Joystick'
Submodule 'libraries/Keyboard' (https://github.com/earlephilhower/Keyboard.git) registered for path 'libraries/HID_Keyboard'
Submodule 'libraries/Mouse' (https://github.com/earlephilhower/Mouse.git) registered for path 'libraries/HID_Mouse'
Submodule 'libraries/LittleFS/lib/littlefs' (https://github.com/littlefs-project/littlefs.git) registered for path 'libraries/LittleFS/lib/littlefs'
Submodule 'libraries/http_parser/lib/http-parser' (https://github.com/nodejs/http-parser.git) registered for path 'libraries/http-parser/lib/http-parser'
Submodule 'ota/uzlib' (https://github.com/pfalcon/uzlib.git) registered for path 'ota/uzlib'
Submodule 'pico-sdk' (https://github.com/raspberrypi/pico-sdk.git) registered for path 'pico-sdk'
Submodule 'tools/libbearssl/bearssl' (https://github.com/earlephilhower/bearssl-esp8266.git) registered for path 'tools/libbearssl/bearssl'
Submodule 'system/pyserial' (https://github.com/pyserial/pyserial.git) registered for path 'tools/pyserial'
Cloning into '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571/ArduinoCore-API'...
remote: Enumerating objects: 1974, done.        
remote: Counting objects: 100% (903/903), done.        
remote: Compressing objects: 100% (338/338), done.        
remote: Total 1974 (delta 590), reused 801 (delta 547), pack-reused 1071 (from 1)        
Receiving objects: 100% (1974/1974), 600.14 KiB | 8.57 MiB/s, done.
Resolving deltas: 100% (1251/1251), done.
Cloning into '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571/lib/tlsf'...
remote: Enumerating objects: 174, done.        
remote: Counting objects: 100% (131/131), done.        
remote: Compressing objects: 100% (55/55), done.        
remote: Total 174 (delta 79), reused 123 (delta 74), pack-reused 43 (from 1)        
Receiving objects: 100% (174/174), 59.79 KiB | 3.32 MiB/s, done.
Resolving deltas: 100% (91/91), done.
Cloning into '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571/libraries/Adafruit_TinyUSB_Arduino'...
remote: Enumerating objects: 6986, done.        
remote: Counting objects: 100% (2132/2132), done.        
remote: Compressing objects: 100% (900/900), done.        
remote: Total 6986 (delta 1400), reused 1621 (delta 1169), pack-reused 4854 (from 1)        
Receiving objects: 100% (6986/6986), 2.26 MiB | 11.22 MiB/s, done.
Resolving deltas: 100% (4215/4215), done.
Cloning into '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571/libraries/AsyncUDP'...
remote: Enumerating objects: 148, done.        
remote: Counting objects: 100% (23/23), done.        
remote: Compressing objects: 100% (18/18), done.        
remote: Total 148 (delta 7), reused 11 (delta 3), pack-reused 125 (from 1)        
Receiving objects: 100% (148/148), 33.38 KiB | 6.68 MiB/s, done.
Resolving deltas: 100% (56/56), done.
Cloning into '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571/libraries/ESP8266SdFat'...
remote: Enumerating objects: 6148, done.        
remote: Counting objects: 100% (3180/3180), done.        
error: RPC failed; curl 18 Transferred a partial file
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
fatal: clone of 'https://github.com/earlephilhower/ESP8266SdFat.git' into submodule path '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571/libraries/ESP8266SdFat' failed
Failed to clone 'libraries/ESP8266SdFat'. Retry scheduled
Cloning into '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571/libraries/FatFS/lib/SPIFTL'...
remote: Enumerating objects: 50, done.        
remote: Counting objects: 100% (50/50), done.        
[... 62 lines cut ...]
[... cut ...]
fatal: clone of 'https://github.com/earlephilhower/ESP8266SdFat.git' into submodule path '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571/libraries/ESP8266SdFat' failed
Failed to clone 'libraries/ESP8266SdFat' a second time, aborting
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'master', 'https://github.com/earlephilhower/arduino-pico.git', '/Users/sstaub/.platformio/.cache/tmp/pkg-installing-xqj6e571']

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
```

maxgerhardt, 2024-09-08:
The git clone is aborting due to an EOF. Seems like unstable internet to me. Try some solutions from https://stackoverflow.com/a/66060728/5296568, like increasing the buffer size, or retrying multiple times..

sstaub, 2024-09-08:
I deleted the platform-arduinopico folder and then start building again, now it works.

sstaub, 2024-09-29:
Annoying, meanwhile it doesn't work again, it happen after 4.0.3
I tried to delete all pico stuff with no luck.

```
sstaub@sstaub1 eOS3 % run -verbose
zsh: command not found: run
sstaub@sstaub1 eOS3 % 
 *  History restored 

sstaub@sstaub1 eOS3 % run -verbose
zsh: command not found: run
sstaub@sstaub1 eOS3 % pio run -verbose
UnknownEnvNamesError: Unknown environment names 'rbose'. Valid names are 'pico'
sstaub@sstaub1 eOS3 % pio run -verbose
UnknownEnvNamesError: Unknown environment names 'rbose'. Valid names are 'pico'
sstaub@sstaub1 eOS3 % pio run
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino; board: rpipico)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0+sha.f7b529a) > Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40003.0+sha.f49d058 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240926 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240926 (14.2.0)
Flash size: 2.00MB
Sketch size: 1.50MB
Filesystem size: 0.50MB
Maximium Sketch size: 1568768 EEPROM start: 0x101ff000 Filesystem start: 0x1017f000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 69 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Ethernet3 @ 1.5.5
|-- SSD1803A_I2C @ 1.0.1
|-- eOS3 @ 0.8.0
Building in release mode
Linking .pio/build/pico/firmware.elf
/Users/sstaub/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /Users/sstaub/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `std':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0xda): undefined reference to `__retarget_lock_init_recursive'
/Users/sstaub/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x124): undefined reference to `__retarget_lock_init_recursive'
/Users/sstaub/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x17c): undefined reference to `__retarget_lock_init_recursive'
/Users/sstaub/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /Users/sstaub/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `__sfp':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:201:(.text+0x1ea): undefined reference to `__retarget_lock_init_recursive'
collect2: error: ld returned 1 exit status
*** [.pio/build/pico/firmware.elf] Error 1
=============================================================== [FAILED] Took 1.18 seconds ===============================================================
sstaub@sstaub1 eOS3 % 
 *  History restored 

sstaub@sstaub1 eOS3 % pio run --verbose
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; framework: arduino; board_build.core: earlephilhower; board_build.filesystem_size: 0.5m; board: rpipico; platform_packages: framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#master; build_flags: -Wall)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0+sha.f7b529a) (git+https://github.com/maxgerhardt/platform-raspberrypi.git) > Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40003.0+sha.f49d058 (git+https://github.com/earlephilhower/arduino-pico.git#master) 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240926 (14.2.0, https://github.com/earlephilhower/pico-quick-toolchain/releases/download/4.0.0/x86_64-apple-darwin20.4.picotool-8a9af99.240926.tar.gz) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240926 (14.2.0, https://github.com/earlephilhower/pico-quick-toolchain/releases/download/4.0.0/x86_64-apple-darwin20.4.arm-none-eabi-8b9df28.240926.tar.gz)
Flash size: 2.00MB
Sketch size: 1.50MB
Filesystem size: 0.50MB
Maximium Sketch size: 1568768 EEPROM start: 0x101ff000 Filesystem start: 0x1017f000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Ignored library /Users/sstaub/.platformio/packages/framework-arduinopico/libraries/Adafruit_TinyUSB_Arduino
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Found 69 compatible libraries
Scanning dependencies...
[... cut ...] ib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `__sfp':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:201:(.text+0x1ea): undefined reference to `__retarget_lock_init_recursive'
collect2: error: ld returned 1 exit status
*** [.pio/build/pico/firmware.elf] Error 1
=============================================================== [FAILED] Took 1.09 seconds ===============================================================
sstaub@sstaub1 eOS3 % 
```

maxgerhardt, 2024-09-29:
```
 - framework-arduinopico @ 1.40003.0+sha.f49d058 (git+https://github.com/earlephilhower/arduino-pico.git#master) 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240926 (14.2.0, https://github.com/earlephilhower/pico-quick-toolchain/releases/download/4.0.0/x86_64-apple-darwin20.4.picotool-8a9af99.240926.tar.gz) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240926 (14.2.0, https://github.com/earlephilhower/pico-quick-toolchain/releases/download/4.0.0/x86_64-apple-darwin20.4.arm-none-eabi-8b9df28.240926.tar.gz)
```
You are using the bleeding edge platform version which now has GCC 14.2.0 with an older Arduino-Pico version, those two are incompatible.

Specify
```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#5e87ae34ca025274df25b3303e9e9cb6c120123c

sstaub, 2024-09-29:
Ok, changing platform work for the moment.

maxgerhardt, 2024-09-29:
I'm expecting the RISC-V branch to be merged into Arduino-Pico any time now, then `arduino-pico.git#master` and the `#develop` of this platform will work together again as normal.

maxgerhardt, 2024-09-30:
We moved back to mainline Arduino-Pico just now. Your old settings should work again.

closed 2024-09-30 by maxgerhardt