maxgerhardt/platform-raspberrypi #76: Pico SDK 2 include path

carlk3, 2024-09-19:
How can I add the SDK header `src/rp2_common/pico_aon_timer/include/pico/aon_timer.h`  to my include path? If I 
```
#include "pico/aon_timer.h"
```
I get 
```
fatal error: pico/aon_timer.h: No such file or directory
```
.

maxgerhardt, 2024-09-22:
The AON Timer component is actually not built by default in Arduino-Pico. So even though you could add the header location to the include path, you would just end up with undefined references. Instead, you have to do something a bit more fancy to also compile the `aon_timer.c` source file.

In your `platformio.ini`, add
```ini
extra_scripts = pre:add_aon.py
```
with `add_on.py` as a new file on the same level as the `platformio.ini` with content
```py
Import("env")
from os.path import join
platform = env.PioPlatform()
FRAMEWORK_DIR = platform.get_package_dir("framework-arduinopico")
env.Append(CPPPATH=[
    join(FRAMEWORK_DIR, "pico-sdk", "src", "rp2_common", "pico_aon_timer", "include")
])
env.BuildSources(
    join("$BUILD_DIR", "PicoAON"),
    join(FRAMEWORK_DIR, "pico-sdk", "src", "rp2_common", "pico_aon_timer")
)
```
Then, a sketch like
```
#include <Arduino.h>
#include "pico/aon_timer.h"
#define LED LED_BUILTIN
void setup() {
  pinMode(LED, OUTPUT);
  struct timespec ts;
  ts.tv_sec = 1727023590; // current unix time
  ts.tv_nsec = 0; 
  aon_timer_start(&ts);
}

void loop() {
  digitalWrite(LED, HIGH);
  delay(1000);
  digitalWrite(LED, LOW);
  delay(1000);
  // print AON time
  struct timespec ts;
  aon_timer_get_time(&ts);
  struct tm tim;
  localtime_r(&ts.tv_sec, &tim);
  char buf[64] = {};
  strftime(buf, sizeof(buf), "%c", &tim);
  Serial.println("Current time: " + String(buf));
}
```
will print
```
Current time: Sun Sep 22 16:46:39 2024
Current time: Sun Sep 22 16:46:41 2024
..
```

maxgerhardt, 2024-09-22:
If you want the AON timer to be compiled and available by default, you have to file an issue with https://github.com/earlephilhower/arduino-pico/issues though.

closed 2024-09-22 by maxgerhardt

carlk3, 2024-09-22:
Thank you! I did get it to work, but it is cumbersome. You script works for Pico, but for Pico 2 I had to add Powman:
```python
Import("env")
from os.path import join
platform = env.PioPlatform()
FRAMEWORK_DIR = platform.get_package_dir("framework-arduinopico")
env.Append(CPPPATH=[
    join(FRAMEWORK_DIR, "pico-sdk", "src", "rp2_common", "pico_aon_timer", "include")
])
# pico-sdk/src/rp2_common/hardware_powman/include/hardware/powman.h
env.Append(CPPPATH=[
    join(FRAMEWORK_DIR, "pico-sdk", "src", "rp2_common", "hardware_powman", "include")
])
env.BuildSources(
    join("$BUILD_DIR", "PicoAON"),
    join(FRAMEWORK_DIR, "pico-sdk", "src", "rp2_common", "pico_aon_timer")
)
# pico-sdk/src/rp2_common/hardware_powman
env.BuildSources(
    join("$BUILD_DIR", "Powman"),
    join(FRAMEWORK_DIR, "pico-sdk", "src", "rp2_common", "hardware_powman")
)
```
However, that won't work for Pico, so I need to have two separate scripts.

maxgerhardt, 2024-09-22:
You can have both scripts in one, to differentiate the RP2040 from the RP2350 you can use
```py
chip = board.get("build.mcu")
if chip == "rp2040":
   pass
elif chip == "rp2350":
   pass
```

carlk3, 2024-09-22:
Great, thanks!