maxgerhardt/platform-raspberrypi #79: Recent change broke RP2350 FastLED build

zackees, 2024-09-27:
These are the same commit histories. I did a git reset to a previous commit and force push. Noticed that one of the exact same commits for different timed builds is now broken.

Working:

https://github.com/FastLED/FastLED/actions/runs/11067480790

Failing:

https://github.com/FastLED/FastLED/actions/runs/11075386507

zackees, 2024-09-27:
This fixes our build for now

```
RPI_PICO2 = Board(
    board_name="rpipico2",
    platform="https://github.com/maxgerhardt/platform-raspberrypi.git#5e87ae34ca025274df25b3303e9e9cb6c120123c",
    platform_needs_install=True,  # Install platform package to get the boards
    platform_packages="framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git",
    framework="arduino",
    board_build_core="earlephilhower",
    board_build_filesystem_size="0.5m",
)
```

erichelgeson, 2024-09-27:
`undefined reference to `__retarget_lock_init_recursive'` I get this same error in my project too, temp pointing back to `5e87ae34ca025274df25b3303e9e9cb6c120123c` is a temp work around.

maxgerhardt, 2024-09-28:
Arduino-Pico did a toolchain upgrade to GCC 14. You cannot use the old git commit hash for the platform and the bleeding edge `arduino-pico.git`. Either keep both old, or both up-to-date.

You might also need to clear your cache: https://github.com/earlephilhower/arduino-pico/pull/2491

zackees, 2024-09-29:
I'm not sure it's a cache thing. The builders failed and they fetch from the url on every run.

nkinnan, 2024-09-29:
Broke esphome as well

nkinnan, 2024-09-29:
@maxgerhardt Would you mind creating a tag or release periodically? Would be better to reference that directly for downstream build versioning. Also, thanks for maintaining this while the... disagreement is ongoing :)

maxgerhardt, 2024-09-29:
Yes, I'm going to create tags for the ones using the previous toolchain version (as per https://github.com/maxgerhardt/platform-raspberrypi/commit/934fe156aee0cf4a5e3ef813d0b23846591fd026, last ones ver v2.2.0 from [earlephilhower/pico-quick-toolchain/releases](https://github.com/earlephilhower/pico-quick-toolchain/releases)) and the ones we use now (v4.0.0).

The same thing can be achieved right now by referencing the commits though: Last version with the old toolchain is 
```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#5e87ae34ca025274df25b3303e9e9cb6c120123c
```
Toolchain updates are always going to happen, as they do now and in the future. Ideally of course, I would look this platform to be in the PIO registry to always hop from synced stable version to stable version. That is not possible as of now.

So, **extra** care must be taken that projects **do not** use bleeding edge versions of both platform and framework (Arduino-Pico), but fixate both versions in place. E.g.., The above commit hash with
```
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#4.0.2
```
should work perfectly fine. If you **don't** touch the `framework-arduinopico` version, but instead use only
```ini
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
```
you should be fine too, because I take care that the latest commit at the end of the day always results in a working build. Right now, the CI for this platform is running perfectly fine, with Arduino-Pico pointing at the `#riscv` branch. I'm expecting this to be merged shortly, so we will be back on mainline Arduino-Pico with RISC-V support.

zackees, 2024-09-29:
Confirmed working.

closed 2024-09-29 by zackees

maxgerhardt, 2024-09-30:
FYI, platform is switched back to mainline Arduino-Pico, latest commit; So, the latest commit of this platform works again with the latest commit of Arduino-Pico *and* has added RISC-V build support ✨.

I'm also thinking about adding backwards-compatibility for building older Arduino-Pico commits, by some type of version detection.. we could select the older toolchain version automatically to still get a successful build. Since I see *major* projects like FastLED and ESPHome getting build failures from not getting the package management exactly right and running into this hickup, this might make snse.

nkinnan, 2024-09-30:
esphome was not previously fixing platform, but is now referencing #5e87ae34ca025274df25b3303e9e9cb6c120123c directly.  In conjunction with https://github.com/earlephilhower/arduino-pico/releases/download/3.9.4/rp2040-3.9.4.zip which is currently two months old. 

Fixing the version is for sure desirable.  I'm happy to continue pointing at #5e87ae34ca025274df25b3303e9e9cb6c120123c but it might be nice to have a specific release or tag to use instead of a commit.  Not having that might be why platform wasn't fixed to a specific version previously (I don't know the history, I just came in to try to fix it now).  The last tag was over 3 years ago: https://github.com/maxgerhardt/platform-raspberrypi/tags

maxgerhardt, 2024-09-30:
Done with [v1.2.0-gcc12](https://github.com/maxgerhardt/platform-raspberrypi/releases/tag/v1.2.0-gcc12).