maxgerhardt/platform-raspberrypi #80: ESPHOME Linking Error

GernotAlthammer, 2024-09-29:
Linking of ESPHome project does show error

**ESPHome:** 2024.9.1

**YAML configuration:**
esphome:
  name: pico-w-1
  friendly_name: pico-w-1

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

**Error-Message:**
Linking .pioenvs/pico-w-1/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `std':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0xda): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x124): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x17c): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `__sfp':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:201:(.text+0x1ea): undefined reference to `__retarget_lock_init_recursive'
collect2: error: ld returned 1 exit status

maxgerhardt, 2024-09-29:
Don't use the bleeding edge platform version with an old Arduino-Pico version. Fixate both in place. Duplicate of #75, #79, https://github.com/earlephilhower/arduino-pico/issues/2502

maxgerhardt, 2024-09-29:
To be specific: 

Replace
```
rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git
```
with
```
rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git#5e87ae34ca025274df25b3303e9e9cb6c120123c
```

closed 2024-09-29 by maxgerhardt

GernotAlthammer, 2024-09-29:
Danke Maximilian. Firmware build ist ohne Fehler durchgelaufen :)

GernotAlthammer, 2024-10-01:
Hi Maximilian,
the ESPHome 2024.9.2 does fix the "behind the scenes stuff" for the users.
See ESPHome RP2040 update and recommendations:
https://esphome.io/components/rp2040.html