maxgerhardt/platform-raspberrypi #83: framework-arduinopico/libraries/lwIP_ESPHost/src/lwIP_ESPHost.cpp:34:35: error: 'ESPHOST_DATA_READY' was not declared in this scope

mathieucarbou, 2024-10-23:
Hello,

I am the maintainer of ESPAsyncWebServer (https://github.com/mathieucarbou/ESPAsyncWebServer).

We have rpi support, with:

```
[env:raspberrypi]
upload_protocol = picotool
; platform = raspberrypi
platform = https://github.com/maxgerhardt/platform-raspberrypi#v1.2.0-gcc12
board = rpipicow
lib_deps = 
  bblanchon/ArduinoJson @ 7.2.0
  khoih-prog/AsyncTCP_RP2040W @ 1.2.0
build_flags = ${env.build_flags}
  -Wno-missing-field-initializers
```

Sadly compilation fails with:

```
Compiling .pio/build/raspberrypi/libeef/WiFi/WiFiClient.cpp.o
Compiling .pio/build/raspberrypi/libeef/WiFi/WiFiClientSecureBearSSL.cpp.o
/Users/mat/.platformio/packages/framework-arduinopico/libraries/lwIP_ESPHost/src/lwIP_ESPHost.cpp: In constructor 'ESPHostLwIP::ESPHostLwIP()':
/Users/mat/.platformio/packages/framework-arduinopico/libraries/lwIP_ESPHost/src/lwIP_ESPHost.cpp:34:35: error: 'ESPHOST_DATA_READY' was not declared in this scope
   34 |     LwipIntfDev<ESPHost>(SS, SPI, ESPHOST_DATA_READY) {
      |                                   ^~~~~~~~~~~~~~~~~~
Compiling .pio/build/raspberrypi/libeef/WiFi/WiFiMulti.cpp.o
*** [.pio/build/raspberrypi/lib749/lwIP_ESPHost/lwIP_ESPHost.cpp.o] Error 1
```

Someone knows how to fix ?

mathieucarbou, 2024-10-23:
Note: same error with `platform = raspberrypi`

maxgerhardt, 2024-10-28:
Will look at it, thanks

maxgerhardt, 2024-10-28:
Is this intended to run on a Pico W? This one uses the lwip_CYW43 library since it has a Crypress CYW43 WiFI module on board. It doesn't have an ESP8266 or ESP32 as its WiFi module. So the fact that lwIP_ESPHost is compiled at all is wrong and you might need `lib_ldf_mode = deep+` if this was wrongly included by some library, e.g. `AsyncTCP_RP2040W`.

mathieucarbou, 2024-10-28:
Yes, pico w support was recently added as a joint effort with @ayushsharma82 for ESP-DASH, ElegantOTA, etc.
We were pointing to this fork and we use `AsyncTCP_RP2040W` which is the only one available.

But weirdly everything was working fine until the recent changes the past weeks / months.

Is it possible that `https://github.com/maxgerhardt/platform-raspberrypi` was pointing to master and it was changed to develop for example (default branch change) ?

maxgerhardt, 2024-10-28:
I can't reproduce your problem all with the latest platform version; That is, git, not `v1.2.0-gcc12` -- this will give you an older Arduino-Pico 3.x core. Not sure why you'd wanna use that.

The Pico W opens the WiFi and serves the HTML wegpage just fine. See repo

https://github.com/maxgerhardt/pio-ESPAsyncWebServer-test

[image]

maxgerhardt, 2024-10-28:
Actually, it even builds when using v1.2.0-gcc12 tag and the there-in default set Arduino-Pico version. So I can double-not reproduce that compile error.

```
Processing rpipicow (platform: https://github.com/maxgerhardt/platform-raspberrypi.git#v1.2.0-gcc12; board: rpipicow; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipicow.html
PLATFORM: Raspberry Pi RP2040 (1.14.0+sha.5e87ae3) > Pico W
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES:
 - framework-arduinopico @ 1.40002.0+sha.a6ab6e1
 - tool-picotool-rp2040-earlephilhower @ 5.120300.240827 (12.3.0)
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 1.50MB
Filesystem size: 0.50MB
Maximium Sketch size: 1568768 EEPROM start: 0x101ff000 Filesystem start: 0x1017f000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 68 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.2.0
|-- AsyncTCP_RP2040W @ 1.2.0
|-- ESPAsyncWebServer @ 3.3.21+sha.5021dd0
|-- WiFi @ 1.0.0
|-- LittleFS @ 0.1.0
|-- WebServer @ 2.0.0
Building in release mode
Retrieving maximum program size .pio\build\rpipicow\firmware.elf
Flash size: 2.00MB
Sketch size: 1.50MB
Filesystem size: 0.50MB
Maximium Sketch size: 1568768 EEPROM start: 0x101ff000 Filesystem start: 0x1017f000 Filesystem end: 0x101ff000
Checking size .pio\build\rpipicow\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  27.7% (used 72728 bytes from 262144 bytes)
Flash: [===       ]  29.5% (used 462724 bytes from 1568768 bytes)
Building .pio\build\rpipicow\firmware.bin
Building .pio\build\rpipicow\firmware.bin.signed
===== [SUCCESS] Took 10.18 seconds =====
```

The wegpage still shows up fine.

Is this reproducable in a CI build? If it's just your local computer that can't build, then delete
* `C:\Users\<user>\.platformio\.cache`
* `C:\Users\<user>\.platformio\platforms\raspberrypi*`
* `C:\Users\<user>\.platformio\packages\framework-arduinopico*`

and build it again.

closed 2024-10-28 by maxgerhardt

mathieucarbou, 2024-10-28:
I really don't understand. 
You example is equivalent as using:

```
lib_ldf_mode = chain
lib_compat_mode = soft
```

which fails on the project.

```
[env:raspberrypi]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow
lib_ldf_mode = chain
lib_compat_mode = soft
lib_deps = 
  bblanchon/ArduinoJson @ 7.2.0
  khoih-prog/AsyncTCP_RP2040W @ 1.2.0
build_flags = ${env.build_flags}
  -Wno-missing-field-initializers
```

`pio run -e raspberrypi` leads to:

```
/Users/mat/.platformio/packages/framework-arduinopico/libraries/lwIP_ESPHost/src/lwIP_ESPHost.cpp: In constructor 'ESPHostLwIP::ESPHostLwIP()':
/Users/mat/.platformio/packages/framework-arduinopico/libraries/lwIP_ESPHost/src/lwIP_ESPHost.cpp:34:35: error: 'ESPHOST_DATA_READY' was not declared in this scope
   34 |     LwipIntfDev<ESPHost>(SS, SPI, ESPHOST_DATA_READY) {
      |                                   ^~~~~~~~~~~~~~~~~~
```

But  setting:

```
lib_ldf_mode = deep+
lib_compat_mode = strict
```

leads to a bunch of link errors:

```
/Users/mat/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/raspberrypi/src/SimpleServer.ino.cpp.o: in function `_ZNSt17_Function_handlerIFvP21AsyncWebServerRequestRN11ArduinoJson8V720PB2211JsonVariantEEZ5setupEUlS1_S5_E17_E9_M_invokeERKSt9_Any_dataOS1_S5_':
SimpleServer.ino.cpp:(.text._ZNSt17_Function_handlerIFvP21AsyncWebServerRequestRN11ArduinoJson8V720PB2211JsonVariantEEZ5setupEUlS1_S5_E17_E9_M_invokeERKSt9_Any_dataOS1_S5_+0x10): undefined reference to `_ZN17AsyncJsonResponseC1Eb'
/Users/mat/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: SimpleServer.ino.cpp:(.text._ZNSt17_Function_handlerIFvP21AsyncWebServerRequestRN11ArduinoJson8V720PB2211JsonVariantEEZ5setupEUlS1_S5_E17_E9_M_invokeERKSt9_Any_dataOS1_S5_+0x34): undefined reference to `_ZN17AsyncJsonResponse9setLengthEv'
/Users/mat/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/raspberrypi/src/SimpleServer.ino.cpp.o: in function `_ZNSt17_Function_handlerIFvP21AsyncWebServerRequestEZ5setupEUlS1_E18_E9_M_invokeERKSt9_Any_dataOS1_':
SimpleServer.ino.cpp:(.text._ZNSt17_Function_handlerIFvP21AsyncWebServerRequestEZ5setupEUlS1_E18_E9_M_invokeERKSt9_Any_dataOS1_+0x10): undefined reference to `_ZN17AsyncJsonResponseC1Eb'
```

maxgerhardt, 2024-10-28:
Interesting, it can compile the captive example just fine but the "Simple server" not, but that should be the same code as I have in my repo, no? Sadly I will have to take a look at this at another time.

mathieucarbou, 2024-10-28:
I am wondering if `deep+` and `#if __has_include("ArduinoJson.h")` are not playing well together...

mathieucarbou, 2024-10-28:
Thanks a lot for you help :-)

Jason2866, 2024-10-28:
@mathieucarbou deep+ is buggy. Try to avoid. If you look in Platformio Core issues will find a lot of "nice" issues

maxgerhardt, 2024-10-28:
This is ultra weird. I can reproduce the bug locally, on the latest platform. When using that code, the WiFi library is detected to have a dependency on all `lwip_xxx` libraries. Then lwip_ESPHost fails. But how come it did not fail in my repo? It also uses WiFi.h (via WebServer.h etc.) Mysterious.

`` 
Dependency Graph
|-- ArduinoJson @ 7.2.0 (License: Unknown, Path: C:\Users\Max\temp\ESPAsyncWebServer\.pio\libdeps\ci-raspberrypi\ArduinoJson)
|-- AsyncTCP_RP2040W @ 1.2.0 (License: LGPL-3.0, Path: C:\Users\Max\temp\ESPAsyncWebServer\.pio\libdeps\ci-raspberrypi\AsyncTCP_RP2040W)
|   |-- WiFi @ 1.0.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\WiFi)
|   |   |-- lwIP-Ethernet @ 1 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet) 
|   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
|   |   |-- Updater @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\Updater)
|   |   |   |-- MD5Builder @ 1.0.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\MD5Builder)
|   |   |   |-- LittleFS @ 0.1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\LittleFS)
|   |   |   |-- PicoOTA @ 1.0.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\PicoOTA)
|   |   |   |   |-- LittleFS @ 0.1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\LittleFS)
|   |   |-- lwIP_CYW43 @ 1 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\lwIP_CYW43)
|   |   |   |-- lwIP-Ethernet @ 1 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet)
|   |   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
|   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
|   |   |-- lwIP_ESPHost @ 1 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\lwIP_ESPHost)
|   |   |   |-- ESPhost @ 1.0.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\ESPHost)
|   |   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
|   |   |   |-- lwIP-Ethernet @ 1 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet)
|   |   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
|   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
|   |   |-- lwIP_WINC1500 @ 1 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\lwIP_WINC1500)
|   |   |   |-- lwIP-Ethernet @ 1 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\lwIP_Ethernet)
|   |   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
|   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
|   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\Users\Max\.platformio\packages\framework-arduinopico\libraries\SPI)
```
this will need some deeper inspection...

maxgerhardt, 2024-10-28:
Found it.

```
lib_ldf_mode = chain

```
breaks building with WiFi library somehow?? The default mode is already "soft". Maybe this globally overrides the LDF mode for **all** libraries, even if they would like to have their own LDF mode.

mathieucarbou, 2024-10-28:
I must say that I like your tag lol!

At least excluding the lib works, as a workaround if some people have the same issue.

Jason2866, 2024-10-28:
> But the WiFi libary doesn't even have a library.json to control the LDF mode. Hmpf.

That's maybe the root of the problem. I remember we had very weird compile issues with Tasmota now and than. After adding a library.json to every lib all the strange issues where gone.