maxgerhardt/platform-raspberrypi #84: Multiple versions of platform-raspberrypi at the same time?

martincerven, 2024-10-23:
There were some breaking changes of [earle's core & SDK2 with micro ros](https://github.com/micro-ROS/micro_ros_platformio/issues/153#issuecomment-2426751782) so I'm using some older version of platform-raspberrypi with:
```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#196d31bbafaf60b84751b1a415d8dca2365debdf
```

But when I want to test something with latest platform-raspberrypi with different pio project, it keeps downloading all the dependencies over again. 

Is there a way to keep different platform versions installed or it will always overwrite dependencies with current project's ones?

maxgerhardt, 2024-10-28:
I think this is an issue with the PlatformIO core, it sees no necessity to keep the old git version when updating to a new git version. This surely saves space, but not in this case.

I don't think this can be solved right now. We would have to switch to stable packages from the PIO registry for that. But integration into mainline PlatformIO is likely blocked for the foreseeable time.

Jason2866, 2024-10-28:
One way would be not providing via git/hash. Making a release zip and providing the URL will help. It still gets reinstalled but it does not download again, it uses the version in cache.