maxgerhardt/platform-raspberrypi #85: How to use earlier version

jhmaloney, 2024-11-19:
I upgraded my PlatformIO package today to version 1.40200.0 (version string from package.json).

I had been running a version from before the 4.0.0 update to the arcuino-pico framework that introduced support for the Raspberry Pi Pico 2 (RP2350) and SDK 2.0.0. I'm sure there are many improvements in that, but a few things in my system got broken by that update so I'd like to temporarily revert until I can fix the issues.

Could you remind me how to build with an earlier version in PlatformIO?

I tried using this line in the platformio.ini environment:

platform = https://github.com/maxgerhardt/platform-raspberrypi.git@3.9.5

but I got this error:

Submodule path 'builder/frameworks/arduino/mbed-core': checked out '66ff9c5ecf64e0ce6360cbf2d8769228bb1294b4'
PackageException: Package version 1.14.0+sha.19e3012 doesn't satisfy requirements 4.0.0 based on PackageMetadata <type=platform name=raspberrypi version=1.14.0+sha.19e3012 spec={'owner': None, 'id': None, 'name': 'platform-raspberrypi', 'requirements': '4.0.0', 'uri': 'git+https://github.com/maxgerhardt/platform-raspberrypi.git'}

What am I doing wrong?

jhmaloney, 2024-11-19:
Another failed attempt that may provide useful info:

Platform Manager: Installing git+https://github.com/maxgerhardt/platform-raspberrypi.git @ 3.7.0
git version 2.45.2
Cloning into '/Users/johnmaloney/.platformio/.cache/tmp/pkg-installing-eahr3ddn'...
remote: Enumerating objects: 192, done.
remote: Counting objects: 100% (192/192), done.
remote: Compressing objects: 100% (165/165), done.
remote: Total 192 (delta 123), reused 52 (delta 8), pack-reused 0 (from 0)
Receiving objects: 100% (192/192), 355.53 KiB | 2.82 MiB/s, done.
Resolving deltas: 100% (123/123), done.
Submodule 'builder/frameworks/arduino/mbed-core' (https://github.com/platformio/builder-framework-arduino-core-mbed.git) registered for path 'builder/frameworks/arduino/mbed-core'
Cloning into '/Users/johnmaloney/.platformio/.cache/tmp/pkg-installing-eahr3ddn/builder/frameworks/arduino/mbed-core'...
remote: Enumerating objects: 43, done.        
remote: Counting objects: 100% (43/43), done.        
remote: Compressing objects: 100% (32/32), done.        
remote: Total 43 (delta 12), reused 33 (delta 10), pack-reused 0 (from 0)        
Receiving objects: 100% (43/43), 13.70 KiB | 4.57 MiB/s, done.
Resolving deltas: 100% (12/12), done.
Submodule path 'builder/frameworks/arduino/mbed-core': checked out '66ff9c5ecf64e0ce6360cbf2d8769228bb1294b4'
PackageException: Package version 1.14.0+sha.19e3012 doesn't satisfy requirements 3.7.0 based on PackageMetadata <type=platform name=raspberrypi version=1.14.0+sha.19e3012 spec={'owner': None, 'id': None, 'name': 'platform-raspberrypi', 'requirements': '3.7.0', 'uri': 'git+https://github.com/maxgerhardt/platform-raspberrypi.git'}

So it seems to be failing when it attempts to checkout the mbed-core submodule?

Thanks for your help!

maxgerhardt, 2024-11-22:
This repo doesn't have any "3.7.0"  or "3.9.5" tags at all. I think what you mean is to set the used Arduino-Pico version to 3.7.0 or 3.9.5 respectively? That's just a *package* within this platform, but not the version of this platform.

You can get a platform version that is pointing at Arduino-Pico 3.9.3 by setting
```ini
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#1ab176cd4d4edd38a156e0c25e12e86ad96b21ef
```

maxgerhardt, 2024-11-27:
Reopen with build log and `platformio.ini` if you still need help.

closed 2024-11-27 by maxgerhardt