maxgerhardt/platform-raspberrypi #88: WiFi reconnect fails after disconnecting and turning off WiFi chip power on Raspberry Pi Pico W

toreis-up, 2024-12-22:
## Description

I'm encountering an issue with the following minimal reproduction code. After disconnecting from WiFi, setting `WiFi.mode(WIFI_OFF)`, and using `digitalWrite(23, LOW)` (GP23: Wi-Fi module power) to turn off the WiFi chip power, the subsequent reconnect attempt fails. The issue specifically arises when trying to reinitialize the WiFiClient and reconnect to the WiFi network.

## Steps to Reproduce

1. Connect to WiFi successfully.
2. Disconnect WiFi using `WiFi.disconnect()` and turn off the WiFi chip power using  `digitalWrite(23, LOW)`.
3. Attempt to reconnect by re-enabling power with `digitalWrite(23, HIGH)` and calling WiFi.begin(ssid, password) again.
4. Try to reinitialize and use a WiFiClient to connect to a server.

## Expected Behavior

The WiFi chip should reconnect to the network, and WiFiClient should be able to establish a connection to the server.

## Actual Behavior

The reconnection process fails, and WiFiClient cannot establish a new connection.

## Reproduction Code

```
#include <Arduino.h>
#include <WiFi.h>

// WiFi Settings
const char* ssid = "SSID";
const char* password = "PASSWORD";

WiFiClient client;

void connectToWiFi();
void reconnectWiFiAndClient();
void disconnectWiFi();

void setup() {
  Serial.begin(115200);

  // Initial WiFi connection
  connectToWiFi();

  // Test communication with WiFiClient
  if (client.connect("example.com", 80)) {
    Serial.println("Connected to server!");
    client.println("GET / HTTP/1.1");
    client.println("Host: example.com");
    client.println("Connection: close");
    client.println();
    delay(1000);
    while (client.available()) {
      String line = client.readStringUntil('\n');
      Serial.println(line);
    }
    client.stop();
  } else {
    Serial.println("Connection to server failed.");
  }

  // WiFi disconnection process
  disconnectWiFi();
  
  // Reconnection process
  reconnectWiFiAndClient();
}

void loop() {
  // No specific tasks in the main loop
}

// Function to connect to WiFi
void connectToWiFi() {
  Serial.print("Connecting to WiFi");
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("\nWiFi connected.");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

// Function to disconnect WiFi and turn off WiFi chip power
void disconnectWiFi() {
  Serial.println("Disconnecting WiFi...");
  client.stop();                   // Stop WiFiClient connection
  WiFi.disconnect();               // Disconnect WiFi
  delay(100);                      // Wait a bit
  WiFi.mode(WIFI_OFF);             // Turn off WiFi mode
  delay(100);                      // Wait a bit
  digitalWrite(23, LOW);           // Turn off WiFi chip power
  Serial.println("WiFi disconnected and power to WiFi chip is off.");
}

// Function to reconnect WiFi and WiFiClient
void reconnectWiFiAndClient() {
  digitalWrite(23, HIGH);          // Turn on WiFi chip power
  delay(100);                      // Wait for stabilization
  Serial.println("Reconnecting WiFi...");
  WiFi.mode(WIFI_STA);             // Set WiFi mode to STA
  connectToWiFi();                 // Reconnect to WiFi

  // Reuse WiFiClient after reconnection
  if (client.connect("example.com", 80)) {
    Serial.println("Reconnected to server!");
    client.println("GET / HTTP/1.1");
    client.println("Host: example.com");
    client.println("Connection: close");
    client.println();
    delay(1000);
    while (client.available()) {
      String line = client.readStringUntil('\n');
      Serial.println(line);
    }
    client.stop();
  } else {
    Serial.println("Reconnection to server failed.");
  }
}
```

## Additional Information

Board: Raspberry Pi Pico W
PlatformIO Framework: Arduino
Library: None

When the code related to WiFiClient is removed, the reconnection process works as expected.
It would be great to know if this is a bug or if there is a missing step in the process for properly reinitializing the WiFi chip and WiFiClient.

Thank you!

maxgerhardt, 2025-02-04:
Very interesting behavior, but this is better discussed in https://github.com/earlephilhower/arduino-pico/issues, since it should not depend on PlatformIO as the build system. (Make sure it also fails when using the Arduino IDE!)

You can also try the latest Arduino-Pico core version in the Arduino IDE (or, after I push an update here).

toreis-up, 2025-02-06:
Thank you for getting back to me.

I tested it with the latest Arduino-Pico core version using the Arduino IDE and reproduced the same issue.
Since it is not specific to PlatformIO, I will report this issue in :)

Thanks again for your guidance!

closed 2025-09-06 by maxgerhardt