maxgerhardt/platform-raspberrypi #90: ENUM hid_report_type_t definition conflict when Adafruit_TinyUSB.h and BluetoothAudio.h are both included

deaddolfin, 2025-01-29:
I wanted to implement audio transmission from Bluetooth to a USB sound card
I wrote the code in VS Code (Platformio)
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
board = rpipicow
build_flags =
-D USE_TINYUSB
-D ARDUINO_RASPBERRY_PI_PICO_W
-D PIO_FRAMEWORK_ARDUINO_ENABLE_BLUETOOTH
lib_deps= https://github.com/pschatzmann/Adafruit_TinyUSB_Arduino.git#Audio

I got it error:
.platformio\packages\framework-arduinopico/pico-sdk/lib/btstack/src/btstack_hid.h:109:35: error: 'HID_REPORT_TYPE_INVALID' conflicts with a previous declaration

The error persists when disabling external libraries.

problematic files files:
.platformio\packages\framework-arduinopico/pico-sdk/lib/btstack/src/btstack_hid.h
.platformio\packages\framework-arduinopico\libraries\Adafruit_TinyUSB_Arduino\src\class\hid\hid.h
possible conflict with
.platformio\packages\framework-arduinopico\pico-sdk\lib\tinyusb\src\class\hid\hid.h

Props3D, 2025-01-31:
This is not an issue with PlatformIO. You're probably better off logging an issue over at https://github.com/earlephilhower/arduino-pico.

However, if you look at some of the bluetooth examples packaged with Arduino-Pico project you'll find a potential work around using a few directives.

maxgerhardt, 2025-02-04:
Does this also happen in the Arduino IDE? If yes, create an issue in arduino-pico itself.

closed 2025-09-06 by maxgerhardt