maxgerhardt/platform-raspberrypi #93: PicoSDK framework doesn't build certian hardware libraries

ProtonNumber, 2025-02-21:
The PicoSDK framework only builds a limited number of the hardware libraries. This seems to be intentional - possibly to help reduce compile time - but there isn't any way to add the remaining libraries without editing picosdk.py.

I'm happy to contribute to this, but I'd like to know what your plans were with regards to this feature.

maxgerhardt, 2025-02-22:
The `framework = picosdk` builder code is not finished at all, PRs welcome.

maxgerhardt, 2025-02-23:
I've only added stuff up until the point where compiling a Blinky program worked and ran, not further. So the other libraries were simply not added yet. Ideally, the PlatformIO builder script should have the same configuration capabilities as the CMake. Some of this was already started (e.g., selecting some options for the math library), but not fully implemented.

I would hope that even if the default is to compile all possible libraries / modules in the Pico SDK, that the link-time-optimization would strip out unused libraries (except their constructor / run on init code of course).

maxgerhardt, 2025-10-17:
Since then, all hardware peripheral libaries have been added.

closed 2025-10-17 by maxgerhardt