maxgerhardt/platform-raspberrypi #108: build option to build filesystem + program into single UF2

indrora, 2025-08-27:
It would be really nice when developing a thing that uses the filesystem to have a "single UF2" shippable option.

Currently, unless I'm dreadfully mistaken, the only way to use the filesystem *and* have preprogrammed files is to 

* Build the application
* Build the littlefs binary
* upload the application
* upload the littlefs binary 

Can we get an option to do both at the same time?

maxgerhardt, 2025-09-06:
I've looked into this, and I'm unable to create a unified UF2 file that has both firmware and filesystem in it. 

Reading https://www.reddit.com/r/raspberrypipico/comments/1khpj8y/comment/mr960tr/ it says I can simply build the firmware.uf2 file (which I already do), build the littelfs.uf2 file (very easy to build form the littlefs.bin file and the `--offset` to the filesystem) and then simply binary-concatenate the two files to get the combined uf2 file.

This is also supported by the format specification https://github.com/microsoft/uf2/tree/d03b585ed780ed51bb0d1e6e8cf233aacb408305?tab=readme-ov-file#file-format which says that it's simply a string of 512 byte blocks.

However, when I do this and push the combined UF2 file onto the BOOTSEL drive, the bootloader on my RP2350 simply aborts after the logical first UF2 chunk has been read and already reboots. It simply doesn't try to read a second UF2 file / chunk.

I'll have to ask Raspberry about this.

maxgerhardt, 2025-09-06:
I have the curious sitation that loading my produced `.uf2` file **works** when using `picotool load` command, but the USB BOOTSEL drive won't accept the file.

maxgerhardt, 2025-09-06:
Temporary workaround in 07679081a49b3052aaa2d541b0770a02d3d31ee1, being a nearly flash-chip size image that has 0xFF padding between firmware and filesystem.

closed 2025-09-10 by maxgerhardt