maxgerhardt/platform-raspberrypi #109: Building on PIO with v5.0.0 needs over 30s

sstaub, 2025-08-28:
Can you have look to this issue I have with v5.0 https://github.com/earlephilhower/arduino-pico/issues/3105

maxgerhardt, 2025-08-29:
I indeed added a "generate disassembly listings" step into the regular firmware build process, but I would not have expected this to add 30 seconds of build time, if I read https://github.com/earlephilhower/arduino-pico/issues/3105#issuecomment-3229053772 right.

Can you locally comment out

https://github.com/maxgerhardt/platform-raspberrypi/blob/0f900eb6d0d77c49d2bb169356d95853a5b637c3/builder/main.py#L405

in `<home dir>/.platformio/platforms/raspberrypi/builders/main.py` and report the time needed for a clean compile? (`pio run -t clean && pio run`)

sstaub, 2025-08-29:
Thank you for that advice. After comment out it works as expected. And yes it takes about 40s more building time.

maxgerhardt, 2025-08-29:
Disabled in d6c42c52acd04ddb6f8e6363215c3f6cb00e1ca2.

closed 2025-08-30 by maxgerhardt